360Controller/360Controller #879: No device found for Rock Candy controller, need someone with Xcode 6.4

mendlo, 2019-05-08:
### Type of Controller
Rock Candy Xbox 360 Controller

### OS Version
macOS Mojave 10.14.4

### Driver Version
I have tried:
- 0.16.11
- 0.16.5
- 0.16
- 0.15_beta5
- 0.15_beta3

### Connection Method
Wired

### Device Name and Info
>Device Name: Rock Candy Wired Controller for PC(XInput)
<br/>Product ID: 0x0152
<br/>Vendor ID: 0x0e6f

### Response of Controller
The device does not light up or vibrate when using macOS. It works well on Windows 10.

### Response in Preference Pane
"No devices found" when the controller is connected via USB 3.0 bus.

### Detailed Explaination of Issue
Because I have a third party controller, the developer clearly explains what to do here:
https://github.com/360Controller/360Controller/blob/master/Readme.md#adding-third-party-controllers

The problem is I have Xcode 10.2. Can someone with Xcode 6.4 or earlier help me out?

Thanks

FranticRain, 2019-05-09:
Here you go.

[360ControllerInstall_0.16.11_Notarized.dmg.zip](https://github.com/360Controller/360Controller/files/3159963/360ControllerInstall_0.16.11_Notarized.dmg.zip)

closed 2019-05-09 by FranticRain

mendlo, 2019-05-09:
Thanks for the quick response, @FranticRain!

360ControllerInstall_0.16.11_Notarized.dmg.zip doesn't seem to be working for me either. I did the following:

1. Uninstalled 0.16.11
2. Rebooted
3. Installed 0.16.11_Notarized from the zip
4. Rebooted
5. In System Preferences -> Xbox 360 Controllers -> Advanced, I enabled the driver

I still get "No devices found."

When I run `sudo kextutil /Library/Extensions/360Controller.kext`, I get:
```
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents
```

If I unplug the Rock Candy controller and plugin an official Microsoft Xbox 360 controller, then the the Microsoft Xbox 360 Ccontroller is detected and works.

FranticRain, 2019-05-09:
Okay, just to make sure that I didn't do something dumb while building it, give this a shot:

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3160113/360ControllerInstall.dmg.zip)

mendlo, 2019-05-09:
It works!!! Multiplayer can commence! Thank you!

FranticRain, 2019-05-09:
Sorry about the mix-up. Glad that it got worked out.