360Controller/360Controller #880: EasySMX ESM-9100 Not working

end3r117, 2019-05-09:
### Type of Controller
EasySMX ESM-9100

### OS Version
MacOS 10.14.3

### Driver Version
360Controller 1.0.0-alpha.2

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: ESM-9100 Wired Game Controller
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

[https://www.amazon.com/gp/product/B06XBX1R55/](url)

### Response of Controller
Lights up for a half a second and then lights turn off and controller is unresponsive and will not turn on.

### Response in Preference Pane
The controller does not appear in the "XBox 360 Controllers" pane.
"No devices found."

### Detailed Explanation of Issue
I believe the above describes it: it just seems the device isn't supported? Do I need to build the driver using the Product ID and Vendor ID? I'm an iOS developer but cloning, editing, and building this into a package looks difficult to me lol. I'm not too proud to admit when I'm out of my depth (and yes I read the dev instructions)!

Any help would be greatly appreciated.

Thank you!

FranticRain, 2019-05-09:
"...Xbox 360/ Xbox One/ ... is not supported"

FranticRain, 2019-05-09:
As mentioned in the amazon listing: "...Xbox 360/ Xbox One/ ... is not supported"

This controller is not supported by official Xbox hardware. As such, its support is outside the scope of this driver. We cannot and will not support this controller.

closed 2019-05-09 by FranticRain

end3r117, 2019-05-09:
Well...that's embarrassing. And a quick return even though I'll feel like a jerk for it. Thanks for the speedy reply!