360Controller/360Controller #883: Analog Inputs Don't Work

ashleyryan15, 2019-05-11:
### Type of Controller
Xbox One
### OS Version
10.14.4
### Driver Version
1.0.0-Alpha2
### Connection Method
Wired
### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:Controller
<br/>Product ID:0x02dd
<br/>Vendor ID:0x045e

### Response of Controller
Guide button illuminates solid when turned on, no vibration but vibration not tested

### Response in Preference Pane
All digital buttons work, but analog sticks and triggers give no response
### Detailed Explaination of Issue
All the digital buttons work as expected, fully normally. Analog inputs give absolutely no response, regardless of modifications of controls.

FranticRain, 2019-05-11:
I can't recreate this issue with the same controller. Would appreciate some confirmation from someone else. Also would like to know if it works in the non-rewrite version, as well as Windows or Xbox.

FranticRain, 2019-10-28:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.