360Controller/360Controller #885: No device detected In Preference Pane despite controller being seen in System Report

Spraypaiint, 2019-05-13:
**Type of Controller**
Xbox One (wired)

**OS Version**
Mojave 10.14.5

**Driver Version**
360Controller 1.0.0-alpha.3 (currently using) and Xone-OSX 1.0.4 (removed)

**Connection Method**
Wired (usb)

**Device Name and Info**
Device Name: 
Product ID: 0x02d1
Vendor ID: 0x045e  (Microsoft Corporation)

**Response of Controller**
The controller lights up when plugged in, but does not vibrate.

**Response in Preference Pane**
The controller does not appear in the Preference Pane.

**Detailed Explanation of Issue**
I originally downloaded Xone-OSX 1.0.4, when I opened it in my settings it detected my controller and was responding to buttons, etc. Although when I tried to use it in game (or anywhere else) It wouldn't work, **which is normal seeing as that driver is outdated**. I then downloaded the new 360 driver and uninstalled Xone-OSX, but now the controller doesn't appear in the preference pane at all. Although the controller is recognized in system support > hardware > USB. Im certain that I have the correct USB cord, so I doubt that's the cause of the issue.

Any help I can get is great! Thank you!
[image]
[image]

FranticRain, 2019-05-13:
Okay, first off, I want to make sure that everything is set up properly, so if you could open up the terminal and run `ls /Library/Extensions` and let me know what the output is, that's the first step.

Next, run `sudo kextutil /Library/Extensions/360Controller.kext` and tell me what that says, after you enter your password.

revehead, 2019-05-17:
Same problem.
Described procedure by @FranticRain returns:

> Notice: /Library/Extensions/360Controller.kext has debug properties set.
> Warnings: 
>     Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
>         ChatPadKeyboardEvents

Thank you.

FranticRain, 2019-05-17:
@revehead I believe your issue in unrelated to the original issue. I also need additional information about your controller. Please make a new issue following the template and the results of your testing so that I can help you.

revehead, 2019-05-17:
great. thanks for quick reply,

Zolthetroll, 2019-05-18:
I have the same problem, though I have no knowledge of code so heres literally everything that came up with each command. 
First one: 
360Controller.kext		HighPointIOP.kext
ACS6x.kext			HighPointRR.kext
ATTOCelerityFC8.kext		PromiseSTEX.kext
ATTOExpressSASHBA2.kext		SoftRAID.kext
ATTOExpressSASRAID2.kext	Wireless360Controller.kext
ArcMSR.kext			WirelessGamingReceiver.kext
CalDigitHDProDrv.kext

Second one: 
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7f916b467630 [0x7fff9c151b30]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7f916b467630 [0x7fff9c151b30]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

does this help?

FranticRain, 2019-05-18:
@Zolthetroll Once again, you have a different issue than OP. You didn't accept the signing certificate attached to this driver during installation. Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI prompt is only present in the Security & Privacy preferences pane for 30 minutes after an alert, such as the one that appeared during installation. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert. You can trigger a kext load by either re-connecting the controller or by running

```sudo kextutil /Library/Extensions/360Controller.kext```

The approval UI is as follows:

[image]

@Spraypaiint I've changed the title of this issue, as your title has invited a lot of off-topic discussion.

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain