360Controller/360Controller #886: PDP stealth not listed in info.plist (verdant green, Midnight Blue)

Wilco2998, 2019-05-14:
### Type of Controller
### Type of Controller
PDP stealth Verdant Green and Midnight Blue

### OS Version
10.14.4

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
USB

### First device Name and Info
Device Name: PDP Wired Controller for Xbox One - Verdant Green
<br/>Product ID: 0x02a1
<br/>Vendor ID: 0x0e6f

### Second device name and info
PDP Wired Controller for Xbox One - Midnight Blue:
<br/>Product ID: 0x02a0
<br/>Vendor ID: 0x0e6f

### Response of Controller
nothing

### Response in Preference Pane
does not appear

### Detailed Explaination of Issue
I saw that the PDP Stealh Violet Spectral & Phantom Black were listed in the info.plist but the controllers listed in this issue are not listed in the info.plist.

I had tried to edit the info.plist but was not able to compile the driver with `./build.sh`

FranticRain, 2019-05-15:
We add these controllers but they don't always work, so far warning that this may not change anything. I've gone ahead and built in into a version of the alpha-3 release as that is most likely to work with these controllers. I'd appreciate knowing whether or not this actually works, so make sure to let me know how it goes. But if it doesn't work, then there isn't much I can do about it. These controller used to work before 10.11, but for some reason they don't anymore.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3180558/360ControllerInstall.dmg.zip)

closed 2019-05-15 by FranticRain

Wilco2998, 2019-05-15:
The Verdant Green is being recognised by the driver and all the buttons are working.
The Midnight Blue did I not test because it's mine brothers controller, I took only the Product and vendor ID so it could be added to the info.plist

I tried also the driver with steam big picture mode, when not playing a game the controller works fine inside the menus.
But what sometimes did happen in steam big picture mode was that the controller was recognised
 multiple times (like in the picture)
[image]