360Controller/360Controller #887: Minor visual bug with a checkbox

PokeCaptain, 2019-05-17:
### Type of Controller
All

### OS Version
10.14.5

### Driver Version
1.0a3

### Connection Method
Wired, but probably N/A

### Device Name and Info
N/A

### Response of Controller
No change/not relevant 

### Response in Preference Pane
See attached 

### Detailed Explaination of Issue
Minor visual bug with the "Enable Driver" checkbox. By default this checkbox is filled in. Clicking it brings up the authentication popup asking "to make changes", aka disable the driver. Clicking cancel here will not disable the driver, however the checkbox will still flip to being unfilled. Vice versa is also true: Clicking the empty box to reenable the driver, then hitting cancel will still fill the checkbox in, even though the driver was not actually enabled. Seems to be a logic issue where the checkbox toggle only detects whether the user clicked the initial box and not if the driver was actually disabled/enabled.
[Screen Recording 2019-05-16 at 11.47.03 PM.zip](https://github.com/360Controller/360Controller/files/3189813/Screen.Recording.2019-05-16.at.11.47.03.PM.zip)

FranticRain, 2019-05-17:
Duplicate of #383. It's on my roadmap for 1.0.0, though.

FranticRain, 2019-05-21:
I took a shot at fixing this issue. Let me know what you think of it.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3200423/360ControllerInstall.dmg.zip)

PokeCaptain, 2019-05-21:
The prefpane handles this much better now, thanks. MacOS booted with the driver disabled after the installation restart FYI. It was able to be enabled of course

FranticRain, 2019-05-21:
Yeah. That’s the expected behavior. The driver shouldn’t be loaded until a device that meets its requirements is actually connected to the computer. You can force load it with the checkbox, if you so desire.

FranticRain, 2019-10-19:
This should be fixed in alpha 4.

closed 2019-10-19 by FranticRain