360Controller/360Controller #889: Controller not working MAC OSX 10.14.5

SteveSoftSoftware, 2019-05-21:
### Type of Controller
Rock Candy Controller for Xbox 36
### OS Version
MAC OSX 10.14.5

### Driver Version
Version 1 Alpha 3, Version 1 Alpha 2, Version 1 Alpha 1

### Connection Method
Wired

### Device Name and Info
 Product ID:	0x011f
  Vendor ID:	0x0e6f
  Version:	1.00
  Serial Number:	32B90ECF
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14700000 / 10
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
It plugs in, lights up once, then no response.
While in the system report screen if I unplug the controller and refresh it, it disappears as expected.  When I plug it back in and refresh, it appears on the USB bus.

### Response in Preference Pane
Nothing at all.

### Detailed Explanation of Issue
It literally does not show up in the device list of the driver.  At one time, a couple years ago (IE a couple MAC OSX versions ago) it did actually work.  I wasn't at that point to use it in my game development adventures.  Now that I am at that point, I can't.  I have an actual Microsoft controller, on the same USB port, with the same cable, has the same result.  So I feel there's a communication failure somewhere.

Thank you.

FranticRain, 2019-05-21:
Can you run this command in your terminal and tell me what the output is? The fact that an official controller isn't working is concerning to me. The Rock Candy one not working is not uncommon.

```
sudo kextutil /Library/Extensions/360Controller.kext
```

SteveSoftSoftware, 2019-05-21:
I will do that when I get back home.  I am at a clients location installing network printers at the moment.

SteveSoftSoftware, 2019-05-21:
Mac-mini:~ stevestevens$ sudo kextutil /Library/Extensions/360Controller.kext
Password:
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/Library/Extensions/47CDA5CA-5002-4F61-954F-563D486152BB.kext
Bundle (/Library/Extensions/VBoxUSB.kext) failed to validate, deleting: /Library/StagedExtensions/Library/Extensions/47CDA5CA-5002-4F61-954F-563D486152BB.kext
Unable to stage kext (/Library/Extensions/VBoxUSB.kext) to secure location.
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/Library/Extensions/B6C01397-C500-4B9F-B800-8CEDF2024AC2.kext
Bundle (/Library/Extensions/VBoxNetFlt.kext) failed to validate, deleting: /Library/StagedExtensions/Library/Extensions/B6C01397-C500-4B9F-B800-8CEDF2024AC2.kext
Unable to stage kext (/Library/Extensions/VBoxNetFlt.kext) to secure location.
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/Library/Extensions/6229F716-EBC7-4579-A706-804C127FAE13.kext
Bundle (/Library/Extensions/VBoxDrv.kext) failed to validate, deleting: /Library/StagedExtensions/Library/Extensions/6229F716-EBC7-4579-A706-804C127FAE13.kext
Unable to stage kext (/Library/Extensions/VBoxDrv.kext) to secure location.
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/Library/Extensions/638574F6-C258-42F5-8BBD-B15150C656B5.kext
Bundle (/Library/Extensions/VBoxNetAdp.kext) failed to validate, deleting: /Library/StagedExtensions/Library/Extensions/638574F6-C258-42F5-8BBD-B15150C656B5.kext
Unable to stage kext (/Library/Extensions/VBoxNetAdp.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7fc6f0d35a10 [0x7fff9b24b8e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7fc6f0d35a10 [0x7fff9b24b8e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents

Mac-mini:~ stevestevens$

So…  it doesn’t have permission!  Yet I get no prompt…...

SteveSoftSoftware, 2019-05-21:
Thank you Apple for making it SO obvious that I had to go to System Preferences->Security & privacy -> general THEN run your terminal command to get a prompt to allow the kext!

So… I got it working with your assistance, thank you so much!  Now I can fly the helicopter I programmed in unity with a joystick instead of the keyboard.  lol!

FranticRain, 2019-05-21:
👍

closed 2019-05-21 by FranticRain