360Controller/360Controller #890: Can't uninstall Driver

lightassassin32, 2019-05-23:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
macOS 10.13.6

### Driver Version
1.0.0-alpha.3
### Connection Method
Wired

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
N/A

### Detailed Explaination of Issue
I can't uninstall the driver and I want to just uninstall the driver as it doesn't work for the Game I want to play

FranticRain, 2019-05-23:
Why can’t you use the uninstall feature in the preference pane? Does it give you an error?

lightassassin32, 2019-05-23:
yes

lightassassin32, 2019-05-23:
is it possible for you to just send me the uninstaller from the download because my download didn't come with one.

lightassassin32, 2019-05-23:
Nevermind, It is gone, It isn't in my system preferences even though I couldn't uninstall it, it would give me this error message every time, can you try making a version that works with Wizard 101, my friend and I would find it way easier to play that game with a controller rather than a mouse and keyboard. Contact me if you'd like to talk about it. thanks!

closed 2019-05-23 by lightassassin32

FranticRain, 2019-05-23:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.