360Controller/360Controller #891: No devices found. 10.14.5

mishkaechoes, 2019-05-25:
### Type of Controller
XBox 360 wireless bluetooth as well as a wired XBox 360 controller

### OS Version
10.14.5

### Driver Version
Xbox 360 Controller Driver
Version 1.0.0-alpha.2

### Connection Method
Wired and wireless with a bluetooth adapter

### Device Name and Info
N/A

### Response of Controller
Controller is flashing green center circle at the moment. I've tried unpairing and repairing a few times.

### Response in Preference Pane
No devices found in the preference pane

### Detailed Explanation of Issue
I've tried 4 different drivers including one off github, alpha 3, alpha 2 and 0.016.6 which was recommend on a youtube video ... Non worked.
I've confirmed security permissions and that the checkbox under Advanced is checked.
Neither of my two controllers appear be detected. Not sure what else to do.
These controllers do work on PC.

FranticRain, 2019-05-26:
You must update your original post with your controller information. A guide on how to do so can be found here: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id.

Controller information is only not required for issue that do not involve controllers. For example, a feature request that isn't specific to any controller, or maybe a typo somewhere in the code or documentation. For all issues individuals have with getting a controller working, controller information is ABSOLUTELY VITAL to solving your problem. Not only does it let us know if your controller is already supported, it also lets us know that the operating system can see the controller and is a crucial step in resolving your issue.

Please notify my when you have updated your original post with your controller information, or let me know if you are unable to find it.

Furthermore, there is no "Bluetooth" Xbox 360 controllers. I'm not sure what you think you are trying to do here.

mishkaechoes, 2019-05-26:
@FranticRain Updated the information.
So earlier today I managed to get it working (or rather it worked) after I randomly pushed buttons while in a game ... the drop down in settings never updated and continued to read No Devices.
After closing the game and opening it up again the controller no longer seems to connect/work.

Hopefully the controller is now clarified in the edited description. Thank you for your help.

FranticRain, 2019-05-26:
I don’t know what version you settled on, but the only version the wireless Xbox 360 controller is going to work with is alpha-3 or later. 0.16.6 is unsupported and ask do not recommend running it. Furthermore, it won’t work with 0.14.5 since 0.14.5 introduces notarization, and 0.16.6 isn’t notarized. So if you aren’t running alpha-3, then you need to upgrade to that version.

However, like I said before, the controller literally can’t connect to the wireless dongle without the driver working, so I’m still perplexed by this issue. Is there a specific game you are trying to play that you are having this issue with? Some games implement their own Xbox controller support and can conflict with the driver, causing these kinds of problems.

But I really want to focus on this wired controller. Figuring out why a wired controller doesn’t work is usually much easier than figuring out why a wireless one doesn’t work. And usually it’s the same problem for both. So I need a couple things:
1. The device information of the wired controller
1. I need you to confirm that this wired controller is a wired controller and does not use some kind of wired adapter. Wireless controllers will always operate wired, whereas wireless Xbox 360 controllers will only communicate wirelessly. 
1. Run this command in the terminal: ‘sudo kextutil /Library/Extensions/360Controller.kext’ and tell me what it says
1. If the controller ever appears to connect to the system, close out of the preference pane if it is open, and then open it again to check and see if the controller is found.

nickg819, 2019-05-26:
> But I really want to focus on this wired controller. Figuring out why a wired controller doesn’t work is usually much easier than figuring out why a wireless one doesn’t work. And usually it’s the same problem for both. So I need a couple things:
> 
> 1. The device information of the wired controller
> 2. I need you to confirm that this wired controller is a wired controller and does not use some kind of wired adapter. Wireless controllers will always operate wired, whereas wireless Xbox 360 controllers will only communicate wirelessly.
> 3. Run this command in the terminal: `sudo kextutil /Library/Extensions/360Controller.kext` and tell me what it says
> 4. If the controller ever appears to connect to the system, close out of the preference pane if it is open, and then open it again to check and see if the controller is found.

I'm having this issue too. I ran the terminal command and it looks like that didn't work. Also tried #4 without any change. Here's the terminal response:

```
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7fbabe9000e0 [0x7fff8d0558e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7fbabe9000e0 [0x7fff8d0558e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents
```
I have an old 360 wired controller I bought years ago from gamestop that used to work. Another is a "PDP Play" wired controller for xbox 360 - 1916GAA is the model number on the controller. Not sure if any of this helps, but figured I'd respond since I'm no longer able to get these USB controllers to work since updating to Mojave 10.14.5

FranticRain, 2019-05-26:
So that error says that you haven't accepted the developer signing certificate yet. I know you said that you already did, but there are certain things that can prevent that button from working the way you may expect it to work. Apple does this as a means of preventing an unauthorized acceptance of that pop-up. If you use any software that impacts the functionality of your mouse or keyboard such as "Karabiner," or if you use some special input devices like a Wacom tablet, they can actually prevent you from successfully pressing this button. So if you can disable any software or unplug any hardware that may cause a problem, you'll need to try to re-approve the driver's signing certificate. That terminal command that I had you run will allow you to approve the kext in a 30 minute window. In your System Preferences, navigate to the "Security & Privacy" pane, and make sure you are on the "General" tab. After running that command in your terminal, the prompt should appear again.

At least I think that's what this means. It's possible that this could also be a notarization error, but I think this is still the error for not accepting the signing certificate.

angelbauts, 2019-07-31:
Same Issu, after Run this command in the terminal: sudo kextutil /Library/Extensions/360Controller.kext

Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain

nickg819, 2019-10-24:
> This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned [...]

Sorry. I think the security settings make sense. I stopped receiving updates for github in email so I just happened to see this one. I know with the most recent OSX updates, they make it nearly impossibly to let security down unless you disable it through the terminal, a few resets, and something something else. I had to do it for another app, and I'm going to try this repo again soon.