360Controller/360Controller #892: Afterglow Xbox One Controller Not Working

stormyknight89, 2019-06-02:
### Type of Controller
Afterglow Xbox One Controller

### OS Version
Mojave 

### Driver Version
10.14.5

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
N/A

### Response of Controller
Respond Like New.

### Response in Preference Pane
Works fine

### Detailed Explaination of Issue
I just bought the Afterglow brand xbox one controller, im on mojave 10.14.5 and using  360controller 1.0.0 alpha 3.   And its not working.   The controller shows up on the preference pane and it responds perfectly in the pane, but as soon as i start a game the controller no longer works.   I was previously using a microsoft controller and worked fine.   Any fix for this?

FranticRain, 2019-06-02:
If it works in the preference pane, then the driver is working as expected, and any other issues are likely issues with the game specifically. Those are not issues that we can fix. However, if you go to the "Advanced" tab of the preference pane and select the "Pretend to be and Xbox 360 Controller" option, it will likely solve your problem.

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain