360Controller/360Controller #893: XOne controller not lighting up on 10.15 beta.

nnvt, 2019-06-04:
### Type of Controller
Xbox One Elite Wired (Official Microsoft)

### OS Version
10.15 beta 1

### Driver Version
1.0.0 alpha 2 and alpha 3

### Connection Method
Wired, Wireless Adapter

### Device Name and Info
Device Name: Controller
<br/>Product ID: 0x02e3
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Vibrates but doesn't light up

### Response in Preference Pane
Controller does not appear in preference pane.

### Detailed Explaination of Issue
Plugging in the controller will make it vibrate but it won't light up as usual. No devices detected in the pref pane and it doesn't work in any other programs either.

Running kextload returns nothing.

Running kextutil returns:

> Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).

NotTheVacuum, 2019-06-04:
I can reproduce with 1.0.0 alpha 3 (as well as whichever older version was bundled w/OneCast). Not entirely unexpected for a developer preview.

Same kextutil output.

FranticRain, 2019-06-04:
Apple frequently breaks the driver in beta releases. It is usually fixed before launch. Furthermore, I don’t have any interest in supporting beta OSes. If it’s still an issue in the full release, then I’ll investigate. It also looks like the driver loads just fine, so it’s almost guaranteed something that Apple broke.

FranticRain, 2019-06-04:
@NotTheVacuum I never realized that the license for 360Controller allowed corporate use. That’s... frustrating.

d235j, 2019-06-04:
@FranticRain banning a field of use makes a software license no longer open source or free software. Therefore such licenses are very highly frowned upon.

GPL requires anyone distributing the software to share their changes, both binary and source under the same license, and extends to use of the code in other software.

FranticRain, 2019-06-04:
@d235j I understand. I’m just a little frustrated that I have spent several thousands of my own dollars, along with countless hours of my own time to support this driver, only to see someone else turn a profit. This is not to lessen the contributions of anyone else who supports or supported this driver. It is a personal complication for me, and should not be a reflection upon anyone else.

NotTheVacuum, 2019-06-04:
Bundled may be the wrong word. It’s clearly a separate install and reasonably credited. The app works without your software. 

But I feel you.

NotTheVacuum, 2019-06-05:
Regarding the driver breaking in betas, I can understand your perspective. However, now would be a good time to review intended changes to the platform, including DriverKit and Kernel Extensions (not to mention the read-only OS volume). It's entirely possible this is just coincidental and future updates to the beta will restore the functionality that's broken today. It's also entirely possible the bug will go missed if you don't report it, or that it's an intentional change and not an incidental break.

FranticRain, 2019-06-05:
You’re 100% right, and I understand your point. But I support this driver in my free time for fun. I own exactly 1 mac. It is my production machine, and I’m not updating it to a beta version of the OS to test this driver. If there is someone who has already installed the beta OS on their machine, I’m sure that they can debug the issues. After all, as @d235j said, this is true open source software. Not Drew source software. And if OneCast feels like it’s something they should be worried about since they rely on this driver, they should be more than happy to contribute to make sure that it works in Catalina. That’s how FOSS work, right?

d235j, 2019-06-05:
IMO beta support, especially for such an early beta, is not a main point of the driver.

If we're closer to October (when macOS Catalina will be released) and it's still broken, investigation would be worthwhile.

offalynne, 2019-07-20:
Looks like Valve has been pointing their customers here too:
https://steamcommunity.com/discussions/forum/2/144513248278519118/#c1732089092459381599

I will be surprised if the new officially-announced OS support for Xbox One controllers (presumably already presented in the beta? I don't know as I can't check, either) doesn't mean this will remain broken come release, or at least leave the inputs inconsistently mapped. Presumably they'd match the existing Bluetooth mapping.

Either way this is likely to break a lot of stuff in Catalina, since multiple widely-used SDL2 device whitelist maps rely on the default configuration of this driver.

NotTheVacuum, 2019-07-20:
Mojave has support for Bluetooth controllers already, although there’s no central button mapping (depends on the app/game). The only reason I use this driver is to plug in my Elite.

offalynne, 2019-07-20:
The Xbox One BT compatible controllers controllers have always worked fine over, I'm referring to USB connection.

offalynne, 2019-07-20:
I only made mention of the Bluetooth connectivity because the input mappings over Bluetooth connectivity don't match those output by this driver (which predates them). A lot of software makes use of the defaults as defined by this driver. I can't confirm that MacOS Catalina intends to implement connectivity over USB but it seems reasonable to assume.

MaddTheSane, 2019-07-20:
The Xbox One BlueTooth controller will be available to the OS X's GameController framework in Catalina. It should be available via IOKit as far back as OS X had BlueTooth.

RayJay-UI, 2019-10-09:
> 并且它仍然存在，那么进行调查是值得的。

macos 10.15 final version，The driver is unavailable.

johny-gog, 2019-10-13:
Doesn't work for me also. Device not detected. :(

FranticRain, 2019-10-19:
This should work now as of alpha 4.

closed 2019-10-19 by FranticRain