360Controller/360Controller #895: New code signer wanted

FranticRain, 2019-06-05:
Hello everyone!

I am no longer comfortable using my developer signing certificate to sign the driver. I plan to continue support of the driver through the month of June. However, no further versions of the driver will be signed by me, starting now.

I am happy to support the driver as a compatibility layer for a dark period of macOS gaming that had terrible controller support. But as I'm seeing more and more applications that continue to rely on the driver rather that building user-space controller support into their app, I realize that I am contributing to the continuation of this dark age. I am not comfortable with this idea and therefore I will stop signing the driver.

Thanks for reading, and thanks for using 360Controller!

d235j, 2019-06-05:
Is there any reason that common frameworks e.g. SDL2, and SDKs e.g. Unity and Unreal, cannot include userspace support for the driver?

d235j, 2019-06-05:
There is also the issue of managing the wireless controller and receiver, which I find it unlikely for apps to do user-space.

FranticRain, 2019-06-05:
@d235j In my opinion, the way forward in wireless is with Bluetooth Xbox One and Bluetooth Dualshock 4 controllers. I'm aware that there are concerns with the number of devices that can be connected at once via Bluetooth. But as macOS ramps up on its official support of these devices via Bluetooth, my interest in supporting wireless 360 controllers wanes.

Deco354, 2019-07-21:
I don't know anything about developing Drivers but I noticed this in Apple's WWDC announcement about driver support in user space. Maybe it's of some use.
https://developer.apple.com/documentation/driverkit

HanSeoulOh, 2019-08-29:
Just saw this and this made me sad :(. Good luck with your next venture!

ForkFaceJones, 2019-10-01:
Just found this and an old 360 controller and AAAAGH 

But still.

Best of luck with everything @FranticRain - thanks for all the work you put in.

DaBlitzStein, 2019-10-17:
Nobody will sign?

FranticRain, 2019-10-27:
@d235j Would you mind enabling the sponsor button functionality since it looks like I'm going to be around for a while longer, looks like.

https://help.github.com/en/github/building-a-strong-community/displaying-a-sponsor-button-in-your-repository#displaying-a-sponsor-button-in-your-repository

j1life1chance, 2019-10-31:
I am willing to sign, although you will have to walk me through what is needed.

FranticRain, 2019-10-31:
@j1life1chance Do you currently have a kext signing certificate? Apple no longer gives them out, to my knowledge.

FranticRain, 2019-12-01:
Since apparently I can't even get the repository owners to do tiny little settings changes for me, I'm going to be ceasing support of 360Controller, effective immediately. It's clear that the owners of this repository have no intention of helping me support this repository, and that this is clearly a dead project that I've been "Weekend At Bernie's"-ing for a number of years.

TyanColte, 2019-12-02:
@FranticRain it had a good run.

maxrothman, 2019-12-15:
@FranticRain if you're having trouble getting support from the repo owners, why not fork?

d235j, 2019-12-15:
@FranticRain I missed your message. You now have admin rights to this repo, and I enabled sponsorship. Sorry for the delay!!

electric-monk, 2019-12-15:
@FranticRain I'd also like to apologise, I set up a filter recently as sometimes github gets a bit noisy so I didn't notice your comment, otherwise I'd have set it up too.

FranticRain, 2020-05-15:
Closing this issue since this driver is soon to be deprecated by Apple is favor of DriverKit. I will be focusing my efforts there.

closed 2020-05-15 by FranticRain