360Controller/360Controller #896: [PDP][Discussion] Anyway to get it work?

thvardhan, 2019-06-06:
Thank you for this amazing driver and the support you are providing. I really appreciate your efforts. I've done some research and figured out that the PDP and PowerA controllers are not supported through this drivers because of some apple code issue.
I would like to ask, is there any alternative to make them work for all those who have these controllers? Maybe like some sort of middleware? or dongle or something? I was thinking of connecting the controller to android and develop an app that can send the button clicks to Mac using wifi or something. To my surprise, the controller does not work on android as well (connected via usb).
is there any workaround?

FranticRain, 2019-06-06:
Does it not work with the 1.0.0 beta versions of the driver? I was under the impression that it had helped resolve the issue. I don’t know for sure though. No one has confirmed or denied.

thvardhan, 2019-06-06:
oh? can you provide me a link? I think I am unable to find the beta version to download.

FranticRain, 2019-06-06:
Sorry. I meant alpha. I forgot my own terms I used.

thvardhan, 2019-06-06:
I am on MacOS Mojave and using the alpha 3 version.
I can confirm that the PDP gamepad does not work. Does not work in the binding menu nor in steam. The light does not turn on as well...

FranticRain, 2019-06-07:
Well first, you would need to plug it in to a Windows PC in order to capture the initialization packets. [I have a guide on how to do this here.](https://vestigl.com/sniffing/index.html) The packets in question will likely start with `0x05 0x20` or `0x04`.

Then you will need to go in and change these lines of code to your new initialization packets:

https://github.com/360Controller/360Controller/blob/5597a3f43113aa0e17ddde010befc7016a046f10/360Controller/_60Controller.cpp#L527-L536

If those packets work for all controllers, then problem solved. I was under the impression that these devices did work with versions of macOS pre-10.11, so I may be wrong on this one. But given the fact that upgrading to the new kext libraries didn't solve this one, my old assumptions appear to be completely wrong. So I think the initialization packets are at least a good place to start on this one.

[If you need some more documentation on what packets mean what in the Xbox One controller protocol, I also have a guide to that here.](https://vestigl.com/packetformat/packetformat.html)

Getting an initialization sequence that properly starts a PDP controller is a crucial first step to making these devices work.

thvardhan, 2019-06-07:
I got a friend's pc and did the packet sniffing. Opening in wireshark, I think I am unable to find packets with 0x05 or 0x04. all I see are 0x00 and 0x01. I am attaching the file here
https://drive.google.com/file/d/1lZZuxyZRLBtGBbwJ5jF5fX6F3H7q74EV/view?usp=sharing

FranticRain, 2019-06-08:
So if you open this up in Wireshark, the part you are looking for is the "Leftover capture data." You'll see that in packet number 37, you the computer sends `0x04 0x20 0x01 0x00` to the controller. The controller then responds on frame 39 with: `0x04 0xf0 0x02 0x3a 0xca ...`. The driver sends a packet starting with `0x01 0x20 ...` in frame 40, and the controller then responds with its name in frame 43 and 45. There's some `0x04` chatter and then at frame 59, we get the main initialization packet that we're familiar with. Starts with `0x05 0x20` as usual. I see the first heartbeat packet in frame 67, which should mean that the controller is good to go by that frame. There's a lot of weird chatter after that, and then we get our first button packet in frame 185. Hopefully that can get you started.

thvardhan, 2019-06-08:
Hey, thanks for the help! I do see the packets now, and I will try this. I am downloading Xcode 6.4 right now and as I have some other project going on, and with my crappy internet, I think it will take me some time to do something. I'll keep you updated on this one. Thanks again :D

FranticRain, 2019-06-08:
You can use a modern version of Xcode if you just fork off of the `modern-rewrite` branch. That’s where all of the “alpha” releases come from, and I’ve built them all with Xcode 10.

fightonb3, 2019-06-27:
Any news about getting these controllers to work with the 1.0.0 alpha? This would be huge!