360Controller/360Controller #897: v1.0.0 alpha 3 failure to boot on Mojave

neutroh, 2019-06-08:
Mojave fails to boot after driver installation and enabling developer Drew Mills in Security & Privacy.

FranticRain, 2019-06-08:
Can you hold Command + V to provide some sort of information that could help us solve the issue instead of ignoring the template and just saying "it doesn't work?"

neutroh, 2019-06-09:
I held down Command + R and reinstalled Mojave. After the reinstall the system booted normally without any data loss. All programs and settings were retained, including Xbox 360 controller software in System Preferences. It seems that the driver was successfully installed since my Xbox One controller now works, and the Mojave installation was repaired during the OS reinstall so that it no longer hangs during boot. 

Is there any specific information I can provide that would help identify the root cause?

FranticRain, 2019-06-09:
I doubt that you can provide any information. Best explanation that I can give is that you had some sort of corrupted internal kext that was used by the driver, causing the panic. But after the re-install, that kext was re-installed over its old version, providing 360Controller with a working base kext.

Since this doesn't seem like it's actually an issue with the driver, and is instead a configuration issue,  I'm going to close it. There isn't a whole lot we can do to make sure that your OS isn't corrupt during an installation process.

closed 2019-06-09 by FranticRain