360Controller/360Controller #898: Update prefpane device list on device connect

FranticRain, 2019-06-09:
**Is your feature request related to a problem? Please describe.**
When a device is connected to the computer while the preference pane is open, the dropdown menu does not re-populate to include that controller.

**Describe the solution you'd like**
Utilize the socket connection system to tell the preference pane to update when the daemon gets a device connection notification.

**Describe alternatives you've considered**
Current workaround is to back out and go back in to the preference pane, to force an update.

**Additional context**
The `modern-rewrite` branch contains a new system for handling the driver enable/disable checkbox. This involves a Bonjour advertised service that has the Daemon operating as a server, with the Preference Pane acting as a client. Utilize this connection on `callbackConnected` events in the Daemon found in `360Daemon.m`. Send a message to the Preference Pane in `Pref360ControlPref.m` `- stream:handleEvent:` that lets the Preference Pane know that it needs to update the controller list.

Will likely require a new format to messages sent back and forth between the Daemon and Pref Pane, since the Pref Pane expects an `OSReturn` and not some indicator of device loading.

FranticRain, 2019-10-24:
This looks like it may be automatically fixed by Catalina? Weird.

medusalix, 2019-10-24:
@FranticRain The list also updates automatically when I connect my Xbox One controller via the dongle, but not when I plug it in over USB. This is on Mojave.