360Controller/360Controller #899: GeForceNow streamer crashes during startup with controller connected

kszumski45, 2019-06-10:
### Type of Controller
Xbox 360

### OS Version
10.14.5

### Driver Version
1.0.0-alpha.3

### Connection Method
Wireless Adapter

### Device Name and Info
>Device Name: Xbox 360 Wireless Receiver for Windows
<br/>Product ID: 0x0719
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Detailed Explaination of Issue
Not sure if it's issue with driver or application but I had no problem using the last version of driver before wireless adapter support removal. Streamer crashes during startup with segmentation fault.
`Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   libGeronimo.dylib             	0x000000010e727f33 XBOX_destroyWirelessDevice + 339
1   libGeronimo.dylib             	0x000000010e727c26 XBOX_addDevices + 5078
2   libGeronimo.dylib             	0x000000010e726826 XBOXDeviceInit + 150
3   libGeronimo.dylib             	0x000000010e7227c5 DEV_DriverInit + 117`

It doesn't matter if driver is enabled or disabled in preference pane, works fine in local games. GeForce doesn't have build in support for that controller.

FranticRain, 2019-06-10:
The driver only talks to the OS and not any particular program. At least not intentionally. Some games do try and bypass the OS and access the driver directly. Its very possible that Nvidia is doing that, and is encountering something they didn't expect with the driver. For example, a game should have ABSOLUTELY NO IDEA whether or not a controller is wired or wireless. It shouldn't matter to a game. The fact that they have a function that explicitly mentions wireless devices is telling. Since the driver works correctly with the OS, I'm going to say that this is a geforce experience bug and not a 360Controller bug. I would take this issue to Nvidia and let them know that we'd be willing to fix the issue if they can tell us what it wrong. But I make a point here to never provide per-game fixes in the driver. Debugging someone else's closed source code is already a pain, especially for free, not to mention the fact that I avoid any sort of hacks for any particular game. Games should support controllers. Controllers shouldn't support games.

kszumski45, 2019-06-10:
Thanks for clarification, when I saw XBOX in functions' names I thought maybe they added support and tried uninstalling the driver but games didn't react to the controller.

ArugulaProtozoa, 2019-08-25:
I had the same problem with Geforce Now but when I uninstalled the driver, it worked and controller functionality works too. Perhaps they have implemented their own driver now and its conflicting?

closed 2019-10-24 by FranticRain