360Controller/360Controller #900: Xbox one X controller hooked up with Macbook

mcooke16, 2019-06-12:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

FranticRain, 2019-06-12:
I'm going to need you to fill out the template before I can even understand what your problem is. Comment again when you've filled out the template.

Defsv, 2019-06-14:
No OP but got similar issue 
Type of Controller

Xbox One , Same with PS4 controller

OS Version

MacOS Catalina Beta 1

Driver Version

1.0.0 Alpha 3

Connection Method

Wired, Bluetooth

Device Name:
Xbox One Controller
Product ID:
0x02ea
Vendor ID:
0x045e  (Microsoft Corporation)
Response of Controller

Vibrate when connected via wire

Response in Preference Pane

Device not found 

Detailed Explaination of Issue

Device not found in preference Pane.
Unity cannot get input information.
Game doesn't register input

FranticRain, 2019-06-14:
@Defsv Make a separate issue, and include the output of running `sudo kextutil /Library/Extensions/360Controller.kext` in the terminal.

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain