360Controller/360Controller #901: not connecting to my wired controller or official wireless adapters.

EthanXYZ, 2019-06-16:
### Type of Controller
Xbox 360

### OS Version
10.6.13
### Driver Version
1.0.0-alpha3
### Connection Method
wired
### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Rock Candy Xbox 360 wired controller
<br/>Product ID: 037-010?
<br/>Vendor ID: ??? pdp

### Response of Controller
nothing happens. it worked before but once I updated to the new version, none of my controllers or wireless adapters are working.
### Response in Preference Pane
"no devices found"
### Detailed Explanation of Issue
As I plug in my Berry Boom Rock Candy Xbox 360 wired controller, the light comes on for just a moment before turning back off. "no devices found". I tried this with my official wireless controller adapters as well. all four of them. with them, the lights kept flashing but alas, "no devices found".

FranticRain, 2019-06-16:
I need a little bit of clarification here. First, you said that your OS version is 10.6.13? I don't think there was a 10.6.13 version of macOS. I assume you mean 10.13.6. And when you say that you "updated to the new version," do you mean of macOS or the driver? What version of macOS/driver were you using before?

EthanXYZ, 2019-06-17:
sorry 10.13.6... haha...

EthanXYZ, 2019-06-17:
the new version of the driver. I believe it's on auto update

FranticRain, 2019-06-17:
There is no auto update built in to the driver, and it definitely wouldn't update you to a beta version. what version of the driver were you running before you updated to alpha3?

EthanXYZ, 2019-06-17:
no idea. but I did NOT update on may 12th or any day after

FranticRain, 2019-06-17:
I would like to figure out how you got updated, but I'll focus on the issue at hand now. Can you run this command in your terminal and let me know what the output is?
```
sudo kextutil /Library/Extensions/360Controller.kext
```
That should let us know if the driver is set up and working correctly.

As to the unexpected update, is there any application that either bundles this driver, or in some way relies on this driver that may have pushed an update to you without your knowledge? The driver installs as part of a PKG and that's generally not a subtle process. So I'd be curious to find out how it got updated.

EthanXYZ, 2019-06-17:
sorry, it sounds like I'm contradicting myself a bit here. I don't remember updating to alpha 3. nobody else uses this computer. I installed the driver late last year. it worked fine until a month or so ago. then this happened.

EthanXYZ, 2019-06-18:
I downloaded it from here on github. I really have no idea how it got updated. unless steam somehow has access to the driver.

FranticRain, 2019-06-18:
Well, either way, I need you to run that command in your terminal to help me determine why it isn't working.

Krutonpalmer3, 2019-06-20:
[image]
I have the exact problem exactly what he said I hope this helps

Krutonpalmer3, 2019-06-20:
[image]

FranticRain, 2019-06-20:
@Krutonpalmer3 Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI prompt is only present in the Security & Privacy preferences pane for 30 minutes after an alert, such as the one that appeared during installation. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert. You can trigger a kext load by by running

```sudo kextutil /Library/Extensions/360Controller.kext```

The approval UI is as follows:

[image]

Krutonpalmer3, 2019-06-20:
omg man your a life saver thanks a lot! that was the problem and it works now

christ776, 2019-06-24:
Hi @FranticRain , I'm also getting the "Personality CFBundleIdentifier differs from containing kext's..." like this 

```Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents

Disabling KextAudit: SIP is off```

But I don't have the "Allow" button showing up on the control panel, is there a way to get around this?

FranticRain, 2019-06-24:
@christ776 That prompt you are seeing is normal, and means that the driver has been loaded correctly. What's different about yours is that you've disabled SIP, so you'll never see an approval prompt. With SIP disabled, you don't need to approve the driver. More likely your device is not detected by the driver. Open a new issue so I can help you further instead of muddling this user support case.

FranticRain, 2019-06-25:
@EthanXYZ Give this version of the driver a shot. May fix your issue. https://github.com/360Controller/360Controller/files/3210105/360ControllerInstall.dmg.zip

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain