360Controller/360Controller #903: XBox 360 wireless controller cannot connect to receiver

TaoLinVT, 2019-06-22:
### Type of Controller
Two Xbox 360 wireless controllers

### OS Version
10.12.6

### Driver Version
1.0.0-alpha.3

### Connection Method
Microsoft wireless receiver is connected to 2011 iMac via a 2.0 USB port

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)

Xbox 360 Wireless Receiver for Windows:

  Product ID:	0x0719
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.00
  Serial Number:	E1977BC0
  Speed:	Up to 12 Mb/sec
  Manufacturer:	©Microsoft
  Location ID:	0xfd140000 / 4
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

I booted up my iMAC. The green light on the receiver turned on.
The wireless controller's guide button lighted up after I pressed it.
The guide button's green light kept flashing.
Then the guide button light changed to circling after I pressed the small sync button.
After a while the lights change back to the flashing mode (all four corners flashing at the same time).
There is no vibration or other signals.
There is no device found on the System-Preference 360 controller panel.
Tried "sudo kextutil /Library/Extensions/360Controller.kext"
Got
"
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents
"
Tried the other wireless controllers. The result is the same.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

There is no device found on the System-Preference 360 controller panel. I also checked the Enable Driver option under Advanced. After that I see 360Daemon running in ActivityMonitor. I tried to connect the controllers to the receiver, but they still cannot connect to the receiver.

### Detailed Explaination of Issue

The Microsoft wireless receiver cannot connect with Xbox 360 wireless controllers. I tried reboot, upgraded OS (from 10.12.4 to 10.12.6), changed the Advanced "Enable Driver", uninstalled/reinstalled 1.0.0-alpha 3, pressed and held the Microsoft receiver button long enough so that its light started to flash, pressed and held the sync button my controller as well. None of them worked.

Really appreciate your help!

FranticRain, 2019-06-22:
Okay so it looks like the driver is loaded correctly and should be running. The "enable" checkbox only corresponds to the wired driver, so that shouldn't have any real impact.

I guess lets start with the a quick sanity check. Can you make sure that the wireless receiver works with Windows? Just want to make sure that everything works, and that the controllers can pair, and then we can find out what is wrong with this macOS configuration.

TaoLinVT, 2019-06-22:
I don't have a windows machine to try that out. The wireless receiver comes with Windows CD and documents (I ordered from Amazon). Is there any other way to verify that?

TaoLinVT, 2019-06-22:
Also I didn't see any allow to run windows pop up, or shown in the system preference (Security & Privacy panel). Is it normal?

TaoLinVT, 2019-06-23:
Resolved it by replacing the 1.0.0 - alpha 3 by 0.16.5. Now both of my wireless controllers worked.

closed 2019-06-23 by TaoLinVT