360Controller/360Controller #904: No devices found in 10.15 beta 2

KriTech, 2019-06-23:
### Type of Controller
Xbox One Elite

### OS Version
MacOS 10.15 Beta 2

### Driver Version
v1.0.0-alpha.3, v0.16.11

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 739
<br/>Vendor ID: 1118

### Response of Controller
It vibrate once.

### Response in Preference Pane
it doesn't appear in the preference pane

### Detailed Explaination of Issue
When I connect the controller it just vibrate once and never appear in the preference pane.

FranticRain, 2019-06-23:
Catalina isn't supported by the driver. Duplicate of #893

closed 2019-06-23 by FranticRain

RayJay-UI, 2019-10-09:
> 驱动程序不支持Catalina。

Does the driver support an official version of macOS 10.15?