360Controller/360Controller #905: Wireless controller not detected

christ776, 2019-06-24:
### Type of Controller
OG Xbox 360 Wireless controller

### OS Version
10.14.3

### Driver Version
alpha-3 pre-release

### Connection Method
Wireless Adapter

### Device Name and Info
USB 3.0 Bus
  Host Controller Driver:	AppleUSBXHCISPT
  PCI Device ID:	0xa12f 
  PCI Revision ID:	0x0031 
  PCI Vendor ID:	0x8086 

USB 3.1 Bus
  Host Controller Driver:	AppleUSBXHCIAR
  PCI Device ID:	0x15d4 
  PCI Revision ID:	0x0002 
  PCI Vendor ID:	0x8086 
  Bus Number:	0x00 

USB 3.1 Bus
  Host Controller Driver:	AppleUSBXHCIAR
  PCI Device ID:	0x15d4 
  PCI Revision ID:	0x0002 
  PCI Vendor ID:	0x8086 
  Bus Number:	0x01 

### Response of Controller
Does not even turn-on.

### Response in Preference Pane
Nothing

### Detailed Explaination of Issue
Wireless dongle does not turn-on and doing it by pressing the button does not make it sync with the XBOX 360 Wireless controller (using the sync button). 
This works if I switch to the old driver 0.16.4 though.

christ776, 2019-06-24:
Running 

```
sudo kextutil /Library/Extensions/360Controller.kext
````

Gives the following output
```
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents

Disabling KextAudit: SIP is off
```

FranticRain, 2019-06-24:
So none of those devices are the wireless dongle, but that shouldn't be important since it supposedly works on 0.16.4. Furthermore, loading 360Controller.kext won't do you much good since that driver isn't used for wireless controllers.

Let's start by just making sure that the dongle is listed in the new version, just to make sure. Please find the actual dongle in your device list, and provide the associated information.

christ776, 2019-06-24:
Oh, here it is!

```
Xbox 360 Wireless Receiver for Windows:

  Product ID:	0x0719
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.00
  Serial Number:	E1C92400
  Speed:	Up to 12 Mb/sec
  Manufacturer:	©Microsoft
  Location ID:	0x14400000 / 10
  Current Available (mA):	500
  Extra Operating Current (mA):	0
```

FranticRain, 2019-06-24:
Okay, that's good. Now, lets see if the console is giving us any useful information. Open up `Console.app` (it's in your `Application/Utilities` folder). Make sure that your device is selected, and then enter "from:kernel" into your search bar. This will filter the messages down to only the ones that come from the kernel. Look for any message that may be coming from the driver. For example, the message may start with "start - ...". Messages should show up when you connect the receiver to the computer.

christ776, 2019-06-24:
How about this information?

[image]

```
Requesting to start advertising for clients 16  with (
        {
        kCBAdditionalAppleMfgAdvertisements =         (
        );
        kCBAdvDataAppleMfgData = <07ff4c00 10020b08>;
        kCBAdvOptionUseFGInterval = 0;
        kCBMsgArgIsPrivilegedDaemon = 1;
    }
)
```

```
advertisingRulesOSX - advertisements: (
        {
        kCBAdvAppleMfgTypeKey = 16;
        kCBAdvDataAppleMfgData = <07ff4c00 10020b08>;
        kCBAdvOptionUseFGInterval = 0;
    }
)
```

FranticRain, 2019-06-24:
That's not it. That's from a process called `wirelessproxd`, not the `kernel` process. Look for it one more time, and if it isn't there, then the driver probably isn't being loaded.

christ776, 2019-06-24:
Nothing shows up that seems to be the driver process, I only see `_sendCommandGated (ep: 3): Sending USB buffer size: 17408 command: <private> [Repeated 5 times]` as something that happens regularly.

I've tried to uninstall the driver too, from the "Advanced" tab and it fails. Maybe it's not installed as it should ?

FranticRain, 2019-06-24:
Well, let's check and make sure that the driver is installed in the first place, then. Navigate to `/Library/Extensions/` and see if there are two files called `Wireless360Controller.kext` and `WirelessGamingReceiver.kext`. You can use the Cmd + Shift + G shortcut in Finder to get to it, if you aren't used to browsing around in your root library directory.

christ776, 2019-06-24:
That explains it. I don't have those files. I'll re-install it

christ776, 2019-06-24:
Re-installed. Now I got this on the Console 

```
Xbox 360 Wireless Receiver for Windows: family specific matching fails
```
several lines like that one

FranticRain, 2019-06-25:
That's also normal. That's the operating system checking the receiver against other drivers before it gets to the 360Controller driver. If there are any logs from the driver, they would be after those lines. There should be some sort of line in there from the driver. For example, if it set up correctly, you would see something like this:
```
start - Transform and roll out (4 interfaces)
```

christ776, 2019-06-25:
I've testing it several times plugging and un-plugging the dongle and the `start - Transform and roll out` message never shows...

So I've tried running 
```
sudo kextutil /Library/Extensions/Wireless360Controller.kex
```

and got this 
```
Notice: /Library/Extensions/Wireless360Controller.kext has debug properties set.
/Library/Extensions/Wireless360Controller.kext - no compatible dependency found for com.apple.kpi.iokit.
/Library/Extensions/Wireless360Controller.kext - no compatible dependency found for com.apple.kpi.libkern.
/Library/Extensions/Wireless360Controller.kext - no compatible dependency found for com.apple.kpi.mach.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.iokit.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.libkern.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.mach.
/Library/Extensions/Wireless360Controller.kext - no compatible dependency found for com.apple.kpi.iokit.
/Library/Extensions/Wireless360Controller.kext - no compatible dependency found for com.apple.kpi.libkern.
/Library/Extensions/Wireless360Controller.kext - no compatible dependency found for com.apple.kpi.mach.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.iokit.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.libkern.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.mach.
/Library/Extensions/Wireless360Controller.kext - no compatible dependency found for com.apple.kpi.iokit.
/Library/Extensions/Wireless360Controller.kext - no compatible dependency found for com.apple.kpi.libkern.
/Library/Extensions/Wireless360Controller.kext - no compatible dependency found for com.apple.kpi.mach.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.iokit.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.libkern.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.mach.
Diagnostics for /Library/Extensions/Wireless360Controller.kext:
Dependency Resolution Failures:
    Only incompatible kexts found for these libraries:
        com.apple.kpi.iokit
        com.apple.kpi.libkern
        com.apple.kpi.mach
    Indirect dependencies can't be resolved:
        com.mice.driver.WirelessGamingReceiver
```

And running 
```sudo kextutil /Library/Extensions/WirelessGamingReceiver.kext```

Yields the following

```
Notice: /Library/Extensions/WirelessGamingReceiver.kext has debug properties set.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.iokit.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.libkern.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.mach.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.iokit.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.libkern.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.mach.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.iokit.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.libkern.
/Library/Extensions/WirelessGamingReceiver.kext - no compatible dependency found for com.apple.kpi.mach.
Diagnostics for /Library/Extensions/WirelessGamingReceiver.kext:
Dependency Resolution Failures:
    Only incompatible kexts found for these libraries:
        com.apple.kpi.iokit
        com.apple.kpi.libkern
        com.apple.kpi.mach
```

FranticRain, 2019-06-25:
Oh that's an easy fix. Install this version of the driver: https://github.com/360Controller/360Controller/files/3210105/360ControllerInstall.dmg.zip

christ776, 2019-06-25:
Success !!! Thank you so much :)

christ776, 2019-06-25:
Note: macOS crashed when I switched off the XBOX controller from the Preferences panel, - after a 30-seconds spinning wheel-. Should I report this crash also?

FranticRain, 2019-06-25:
There's been some discussion here: https://github.com/360Controller/360Controller/issues/809

Should probably concentrate all of the information there.

closed 2019-06-25 by christ776