360Controller/360Controller #907: No devices found. - macOS Mojave 10.14.3

benzcooper, 2019-06-26:
### Type of Controller
Xbox 360

### OS Version
10.14.3

### Driver Version
Version 1.0.0-alpha.3

### Connection Method
Wired

### Device Name and Info
Vendor-Specific Device:

  Product ID:	0x028f
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.06
  Speed:	Up to 12 Mb/sec
  Location ID:	0x14141000 / 12
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

### Response of Controller
Full ring will flash when I hit the guide button. Keeps flashing regularily.

### Response in Preference Pane
No devices found

### Detailed Explaination of Issue
I ran `sudo kextutil /Library/Extensions/360Controller.kext` as per advice on the No devices found. 10.14.5 #891 and got this.

> Notice: /Library/Extensions/360Controller.kext has debug properties set.
> Warnings: 
>     Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
>         ChatPadKeyboardEvents

I don't have any other devices (Wacom, etc) installed on this computer.

Any advice?

Thanks very much in advance.

FranticRain, 2019-06-26:
Just out of curiosity, give this build a shot: https://github.com/360Controller/360Controller/files/3210105/360ControllerInstall.dmg.zip

benzcooper, 2019-06-26:
Thanks for the attempt. Unfortunately, no change. 

I realized after I posted that one thing that could be messing it up is the touchbar on my laptop. I have a MacBook Pro (15-inch, 2017). Perhaps that could be causing a problem?

FranticRain, 2019-06-26:
I don't know exactly what the problem is, but the driver appears to be acting correctly and is loaded correctly. It should pair to your controller. I would like to make sure that your controller works properly. Can you confirm that this controller works with Windows or an Xbox console?

benzcooper, 2019-06-26:
Weird... It definitely works with Xbox console.

FranticRain, 2019-06-26:
Okay, so this is where the debugging is going to get tougher. I need you to open up `Console.app` located in your `Applications/Utilities` folder. Make sure that your device is selected in the left hand side panel. Then you want to filter all messages from the console down by using the search bar. If you enter "process:kernel" it will limit all logs to messages that are tagged with the kernel process, which will include the driver. Now, when you connect the controller to the computer, you should see some messages show up. Notably, you will want to look for ones that start with "start - ". So, for example, you may see "start - Failed super start method." Let me know if you see any messages.

amoswazana, 2019-06-26:
Same problem here. This is what is shown in the console:
default	02:02:27.135723 +0300	kernel	Controller: family specific matching fails
default	02:02:27.135734 +0300	kernel	Controller: family specific matching fails
default	02:02:27.135738 +0300	kernel	Controller: family specific matching fails
default	02:02:27.138744 +0300	System Preferences	No devices found? :( device count==0, i==0

benzcooper, 2019-06-26:
Here's what I got: 

> default	16:20:32.000836 -0700	kernel	Starting poll type 4
> default	16:20:32.000843 -0700	kernel	Restarting poll type 4
> default	16:20:32.001325 -0700	kernel	Command 0x1100 completed. inCount: 2 data:0x0
> default	16:20:32.002326 -0700	kernel	Command 0x1200 completed. inCount: 2 data:0x0
> default	16:20:32.003353 -0700	kernel	Command 0x1300 completed. inCount: 2 data:0x0
> default	16:20:32.005293 -0700	kernel	Command 0x2 completed. inCount: 2 data:0x6
> default	16:20:32.007221 -0700	kernel	Command 0x1 completed. inCount: 2 data:0x1101
> default	16:20:32.007358 -0700	kernel	Command 0x16 completed. inCount: 2 data:0xc0
> default	16:20:32.007464 -0700	kernel	Command 0xa completed. inCount: 2 data:0xf945
> default	16:20:32.010249 -0700	kernel	Command 0xb completed. inCount: 2 data:0xfa0f
> default	16:20:32.013056 -0700	kernel	Command 0x12 completed. inCount: 2 data:0xe2
> default	16:20:32.013187 -0700	kernel	Command 0x13 completed. inCount: 2 data:0xffff
> default	16:20:32.013339 -0700	kernel	Command 0xf completed. inCount: 2 data:0x166e
> default	16:20:32.013449 -0700	kernel	Command 0x10 completed. inCount: 2 data:0x190d
> default	16:20:32.013465 -0700	kernel	SmartBattery: finished polling type 4
> default	16:20:32.404662 -0700	kernel	failed to lookup target storage class: 2
> default	16:20:32.430985 -0700	kernel	failed to lookup target storage class: 2
> default	16:20:32.431840 -0700	kernel	failed to lookup target storage class: 2
> default	16:20:32.898702 -0700	kernel	failed to lookup target storage class: 2
> default	16:20:32.898767 -0700	kernel	Sandbox: 1 duplicate report for contextstored deny(1) file-write-data /private/var/root/Library/Preferences/com.apple.contextstored.plist
> default	16:20:32.900139 -0700	kernel	failed to lookup target storage class: 2
> default	16:20:32.904415 -0700	kernel	failed to lookup target storage class: 2
> error	16:20:57.025513 -0700	kernel	Sandbox: QuickLookSatelli(1103) deny(1) mach-lookup com.apple.powerlog.plxpclogger.xpc

benzcooper, 2019-06-26:
I just tried unplugging the controller and plugging it back in. As soon as I plugged it in, I got:

> default	16:23:47.947477 -0700	kernel	Xbox360Peripheral::probe(IOUSBHostDevice)
> default	16:23:47.947481 -0700	kernel	Xbox360Peripheral::start(IOUSBHostDevice) <1>
> default	16:23:47.947525 -0700	kernel	Xbox360Peripheral::start(IOUSBHostDevice) <1> failed

Then I tried to press the guide button on the controller and nothing else happened. A few moments later, it ran through essentially the same sequence as above.

FranticRain, 2019-06-26:
@amoswazana Make a new issue, follow the template, and post all of the debugging you've done.

@benzcooper This implies that the IOUSBHostDevice kext failed. Are you using a genuine mac or a hackintosh?

benzcooper, 2019-06-27:
Genuine Mac : MacBook Pro (15-inch, 2017) . I thought the touchbar might have messed with it somehow. Should I mod the kext?

(My other computer is a hackintosh and I haven't attempted to install it on there yet.)

FranticRain, 2019-06-27:
The touchbar should have absolutely nothing to do with this. It isn't a part of the USB stack and it doesn't matter at all. What does matter is that you are using some sort of USB-C connection in this process since you are using a 2017 rMBP. Some adapters have proven to just not work with the driver. I don't know what kind of adapter you are using, but I think that *might* be the problem. It's hard to say, because the driver should be printing out some log if it was the thing that failed.

It's really hard to say that the driver is at fault here when there are plenty of users that use 360 controllers just fine on Mojave with this version, so I'll admit, I am looking for something on your system that is at fault. I would say 98% of 360Controller issues are someone failing to set up something right, but we've run through all of the checks and you are set up correctly, so it isn't that. But without any of the debug logs from the driver, I don't know how to proceed.

benzcooper, 2019-06-27:
Right. That makes sense.

I'm using a USB hub connected to one of the USB-C ports because the laptop only has USB-C _(grumble grumble)_ so, maybe there is some funkiness happening there? That hub has been known to be a little funny on occasion. I can find someone to borrow a hub from and try to eliminate that potential weak point. Anything else I should try in the meantime?

FranticRain, 2019-06-27:
I honestly can't think of anything else, unless a debug log for the failure reason showed up in your console magically.

manyahin, 2019-06-28:
I updated my Macbook Pro 2015 to last Mac OS (Mojave, 10.14.5) and get the same problem. With a previous version (Sierra?) it was working fine.

FranticRain, 2019-06-28:
@manyahin Make a new issue and list all of the debugging steps that you have taken.

benzcooper, 2019-07-03:
So, I tried this again with a different USB hub connected through USB-C. Two different controllers with the same result.

`default	14:52:06.710912 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.710929 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.710972 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.711095 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.711221 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.711875 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.711918 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.711948 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712097 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712125 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712164 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712310 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712317 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712324 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712335 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712343 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712354 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712365 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712373 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712384 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712390 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712396 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712403 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712411 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712420 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712427 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712433 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712440 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712446 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712453 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712459 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712466 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712473 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712479 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712486 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712492 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712499 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712503 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712509 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712516 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712523 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712530 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712536 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712543 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712550 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712556 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712563 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712568 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712573 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712578 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712584 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712591 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712597 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712604 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712611 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712618 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712624 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712630 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712636 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712643 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712649 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712656 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712662 -0700	kernel	IOUSBHostDevice: family specific matching fails
[... cut ...]
default	14:52:06.712701 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712707 -0700	kernel	IOUSBHostDevice: family specific matching fails
default	14:52:06.712743 -0700	kernel	Xbox360Peripheral::probe(IOUSBHostDevice)
default	14:52:06.712751 -0700	kernel	Xbox360Peripheral::start(IOUSBHostDevice) <1>
default	14:52:06.712782 -0700	kernel	start - Unknown device release 0106
default	14:52:06.712785 -0700	kernel	start - Unable to find the interface.
default	14:52:06.712796 -0700	kernel	Xbox360Peripheral::start(IOUSBHostDevice) <1> failed`

L0VEFIRE, 2019-10-10:
I can tell you right now, your HUB is the issue. Those things do nothing but cause trouble and also degrade you peripheral overall performance.

**(1)** Go get a "USB-C to Micro USB" cable (assuming your using a xbox one controller)
_[](https://www.amazon.com/AmazonBasics-USB-Type-C-Micro-B-Cable/dp/B01GGKYE4U/ref=asc_df_B01LONQ7R6/?tag=hyprod-20&linkCode=df0&hvadid=167140037427&hvpos=1o2&hvnetw=g&hvrand=14432955992272329817&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=&hvtargid=pla-299610941471&th=1)_ 
**(2)** Then install driver 0.16.10 

I bet you it will work. (as long as you never attempted to use bluetooth with the controller ever)

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain