360Controller/360Controller #908: Wont Work with my Macbook

yungmath, 2019-06-27:
### Type of Controller
Xbox One

### OS Version
10.14.2

### Driver Version
0.16.11

### Connection Method
Wired Controller, Regular Controller with Micro USB Cable, Tried everything

### Device Name and Info
N/A (i think?)

### Response of Controller
Sometimes the Light will blink and the controller will vibrate for 1 second right when i plug the cord in but thats all

### Response in Preference Pane
no connection shows up just says no devices found

### Detailed Explaination of Issue
Been trying to connect my xbox one controller for months i cant figure it out i really want to use it for fortnite cause my macbook runs it way better than my xbox, please help

i've tried bluetooth, other drivers, etc.

FranticRain, 2019-06-27:
You must update your original post with your controller information. A guide on how to do so can be found here: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id.

Controller information is only not required for issue that do not involve controllers. For example, a feature request that isn't specific to any controller, or maybe a typo somewhere in the code or documentation. For all issues individuals have with getting a controller working, controller information is ABSOLUTELY VITAL to solving your problem. Not only does it let us know if your controller is already supported, it also lets us know that the operating system can see the controller and is a crucial step in resolving your issue.

Please notify my when you have updated your original post with your controller information, or let me know if you are unable to find it.

Also, the driver isn't needed for Bluetooth connections, so that should have worked without the driver.

L0VEFIRE, 2019-10-10:
The moment you use bluetooth with xbox controller on a Mac it breaks the ability to use it in a wired connection again with these drivers. My suggestion to you is to make a new partition on your Mac (size 50GB) and install Mojave. install the driver and use a USB-C to Micro USB cable (NO HUBS OR ADAPTERS). and it should work fine.

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain