360Controller/360Controller #911: Xbox One controller not found on Mac OS Mojave 10.14.5

ashrod98, 2019-06-28:
### Type of Controller
Xbox One

### OS Version
MacOS Mojave 10.14.5

### Driver Version
360Controller 1.0.0 alpha.3

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 0x02dd
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Vibrates for a moment immediately after connecting it to my mac. There are no lights nor other indicator that it is connected other than the fact that it shows up under USB devices in my system report.

### Response in Preference Pane
Preference Pane does not list my device. The only item in the drop-down menu is No Devices Found.

### Detailed Explanation of Issue
I recently downloaded and installed the most recent driver (alpha 3) onto my mac. Twice, just in case something went wrong. When I connect my controller via micro USB cable to my mac and it vibrates for just a second, but the controller never shows up in system preferences. It does appear as a USB device on my system report. This is the name of the file from which I installed the driver: 360ControllerInstall_1.0.0-alpha.3 16-59-42-537.dmg (Got this from my trash.)

[image]

FranticRain, 2019-06-28:
Can you go into your terminal and run this command, and let me know what the system says?
```
sudo kextutil /Library/Extensions/360Controller.kext
```

kevzettler, 2019-06-29:
@FranticRain I have the same setup as @ashrod98 heres my terminal output

```
$ sudo kextutil /Library/Extensions/360Controller.kext
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7fddf8d045c0 [0x7fff8dcd08f0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7fddf8d045c0 [0x7fff8dcd08f0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents
```

FranticRain, 2019-06-29:
@kevzettler Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI prompt is only present in the Security & Privacy preferences pane for 30 minutes after an alert, such as the one that appeared during installation. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert. You can trigger a kext load by either re-connecting the controller or by running

```sudo kextutil /Library/Extensions/360Controller.kext```

The approval UI is as follows:

[image]

kevzettler, 2019-06-30:
Thanks @FranticRain that resolved the issue for me. Controller is now detected

FranticRain, 2019-06-30:
@kevzettler Apple, in their infinite wisdom, decided to only show the prompt once for any developer. So if you've ever tried to load any software and missed the prompt even once, then you'll never see a pop-up ever again. Just this little "allow" button in your system preferences. It wouldn't surprise me if it just got covered up or missed the one time it showed up and then you just never got a chance to see it ever again.

ashrod98, 2019-07-01:
Thanks! It works for me as well

kawaiicute, 2019-07-04:
[image]

i have this

chumbaker, 2019-07-15:
> Can you go into your terminal and run this command, and let me know what the system says?
> 
> ```
> sudo kextutil /Library/Extensions/360Controller.kext
> ```

Hi! First post here, so sorry if this isn't the right etiquette.
I'm also running 10.14.5, and having the same issues as op. However, when I put that command into terminal, it gave me this:

Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents
      
It's not my controller, laptop, or cord, and I already allowed the program to make changes. I'm really at a loss here :-(

itskando, 2019-07-21:
> When I put that command into terminal, it gave me this:
>
> Error copying bundle during staging: -1
> Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
> Notice: [...]

+1

DanSpicyTaco, 2019-08-03:
> when I put that command into terminal, it gave me this:
>
> Error copying bundle during staging: -1
> Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
> Notice: [...]

I had the same issue and Fixed it by clearing the kextcache:
`sudo kextcache --clear-staging`

And then running the command again:
`sudo kextutil /Library/Extensions/360Controller.kext`

The output for that was:
`Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
kext file:///Library/StagedExtensions/System/Library/Extensions/1085F424-AEF4-41D6-85F1-EB2ECC7843.kext/ is in hash exception list, allowing to load
kext file:///Library/StagedExtensions/System/Library/Extensions/8A667711-5478-489B-A6A2-777DCDB1BB.kext/ is in hash exception list, allowing to load
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents`

Then, I unplugged the controller and inserted it again and it seems to work now! (I'm not entirely sure why though, but it worked for me)

**Type of Controller**
Xbox One (Wired)

**OS Version**
MacOS Mojave 10.14.5

**Driver Version**
360Controller 1.0.0 alpha.3

romansand, 2019-08-08:
Hey guys,

I have the same issue.
MacOS isn't detecting a xbox controller.

The output of:
`sudo kextutil /Library/Extensions/360Controller.kext`
Is
`Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents`

I've tried alpha.2 and alpha.3 versions, but it didn't prompt a Security approval during the installation.
Any ideas/suggestions?

**Type of Controller**
Xbox One (Wired)

**OS Version**
MacOS Mojave 10.14.5

**Driver Version**
360Controller 1.0.0 alpha.3

coal, 2019-08-20:
Same issue as above.

CMP:~ coal$ sudo kextutil /Library/Extensions/360Controller.kext
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

CMP:~ coal$ sudo kextutil /Library/Extensions/360Controller.kext
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

Type of Controller
Xbox One (Wireless)

OS Version
MacOS Mojave 10.14.5

Driver Version
360Controller 1.0.0 alpha.3

bamode, 2019-08-30:
I have the same issue and OS version as Coal above. Can confirm that the exact same issue and output from kextutil exists for wired Xbox One controller. Mojave 10.14.6 contained an update to the USB drivers that is probably the root cause of the issue. Running kextutil at maximum verbosity confirms that the machine seems to believe that the kext is loaded successfully, though it clearly is not. Full output reproduced below.

`Kext user-space log filter changed from 0xff2 to 0xfff.
Kext kernel-space log filter changed from 0xff2 to 0xfff.
Kext library architecture set to x86_64.
Can't stat /AppleInternal - No such file or directory.
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext library recording diagnostics for: validation authentication dependencies warnings.
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents

/Library/Extensions/360Controller.kext appears to be loadable (not including linkage for on-disk libraries).
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Loading /Library/Extensions/360Controller.kext.
Reading load info for 15 kexts.
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
"com.apple.kpi.bsd" = {
    "OSBundleWiredSize" = 40528
    "CFBundleIdentifier" = "com.apple.kpi.bsd"
    "OSBundleUUID" = <4e953e56 65c84d2e 94e10727 512d23a2>
    "OSBundleLoadAddress" = -551890714624
    "OSBundleLoadTag" = 1
    "OSBundleStarted" = true
    "OSBundleCompatibleVersion" = "8.0.0b1"
    "OSKernelResource" = true
    "CFBundleVersion" = "18.7.0"
    "OSBundlePath" =
        "/System/Library/Extensions/System.kext/PlugIns/BSDKernel.kext"
    "OSBundleRetainCount" = 109
    "OSBundlePrelinked" = true
    "OSBundleIsInterface" = true
    "OSBundleLoadSize" = 40528
}
"com.apple.iokit.IOPCIFamily" = {
    "OSBundleWiredSize" = 221184
    "CFBundleIdentifier" = "com.apple.iokit.IOPCIFamily"
    "OSBundleStarted" = true
    "OSBundleCPUType" = 16777223
    "OSBundleCompatibleVersion" = "1.0.0b1"
    "OSBundleUUID" = <202f4f5c 5e3c35a9 a7adf85b 0e3cd199>
    "OSKernelResource" = false
    "CFBundleVersion" = "2.9"
    "OSBundleLoadTag" = 13
    "OSBundleLoadAddress" = -551890104320
    "OSBundleDependencies" = (
        5
[... 337 lines cut ...]
(kernel) Loading kext com.mice.driver.Xbox360Controller.
(kernel) Kext com.apple.iokit.IOHIDFamily is already loaded.
(kernel) Kext com.apple.iokit.IOHIDFamily sending 4 personalities to the IOCatalogue and starting matching.
(kernel) Kext com.apple.kpi.mach is already loaded.
(kernel) Kext com.apple.iokit.IOUSBHostFamily is already loaded.
(kernel) Kext com.apple.iokit.IOUSBHostFamily sending 34 personalities to the IOCatalogue and starting matching.
(kernel) Kext com.apple.kpi.iokit is already loaded.
(kernel) Kext com.apple.kpi.libkern is already loaded.
(kernel) Kext com.apple.kpi.bsd is already loaded.
(kernel) Allocated link buffer for kext com.mice.driver.Xbox360Controller at 0xffffff7f88419000 (90112 bytes).
(kernel) Kext com.mice.driver.Xbox360Controller executable loaded; 22 pages at 0xffffff7f83c19000 (load tag 174).
(kernel) Kext com.mice.driver.Xbox360Controller calling module start function.
(kernel) Kext com.mice.driver.Xbox360Controller registered class ChatPadKeyboardClass.
(kernel) Kext com.mice.driver.Xbox360Controller has IOService subclass ChatPadKeyboardClass; enabling autounload.
(kernel) Kext com.mice.driver.Xbox360Controller registered class Xbox360Peripheral.
(kernel) Kext com.mice.driver.Xbox360Controller registered class Xbox360ControllerClass.
(kernel) Kext com.mice.driver.Xbox360Controller registered class Xbox360Pretend360Class.
(kernel) Kext com.mice.driver.Xbox360Controller registered class XboxOriginalControllerClass.
(kernel) Kext com.mice.driver.Xbox360Controller registered class XboxOneControllerClass.
(kernel) Kext com.mice.driver.Xbox360Controller registered class XboxOnePretend360Class.
(kernel) Kext com.mice.driver.Xbox360Controller is now started.
(kernel) Kext com.mice.driver.Xbox360Controller sending 221 personalities to the IOCatalogue and starting matching.
(kernel) Kext com.mice.driver.Xbox360Controller loaded.
Successfully loaded /Library/StagedExtensions/Library/Extensions/360Controller.kext.
/Library/Extensions/360Controller.kext successfully loaded (or already loaded).`

heran99, 2019-09-06:
+1 same problem here:

> Error copying bundle during staging: -1
> Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
> Notice: /Library/Extensions/360Controller.kext has debug properties set. [...]

**Type of Controller**
Xbox One (Wireless)

**OS Version**
MacOS Mojave 10.14.6

**Driver Version**
360Controller 1.0.0 alpha.3

drizzylga1151, 2019-09-17:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Can't open CFBundle for /System/Library/Extensions/roller.kext.
Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 0.
Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 0.
Kext with invalid signatured (-67054) allowed: <OSKext 0x7f9a28442bc0 [0x7fffa9dc18e0]> { URL = "file:///System/Library/Extensions/IOUSBHostFamily.kext/", ID = "com.apple.iokit.IOUSBHostFamily" }
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 0.

FranticRain, 2019-10-24:
Okay, this issue is a mess of random garbage that I can't individually address. If any of you still have an issue, please make a new issue following the template and include the output from kextutil.

closed 2019-10-24 by FranticRain

vrdevelopersco, 2019-11-08:
seems doing

`sudo kextcache --clear-staging`

then doing

`kextutil /Library/Extensions/360Controller.kext`

does the job even if console throws error, because after completing command I just unplug and replug controller and it's done