360Controller/360Controller #914: Xbox one controller not found on Mac OS Catalina (0.15 Beta)

ra2630, 2019-07-05:
### Type of Controller
Xbox One Wired controller

### OS Version
Mac OS Catalina (0.15 Beta)

### Driver Version
360Controller 1.0.0-alpha.3

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: XB1 Classic Controller
<br/>Product ID: 0x581a
<br/>Vendor ID: 0x24c6

### Response of Controller
No response from controller

### Response in Preference Pane
The controller does not appear in the preference pane

### Detailed Explaination of Issue
I downloaded and installed the most recent driver (alpha 3) on my mac. When I connect my controller via micro USB cable to my mac nothing happens, but the controller never shows up in system preferences. It does appear as a USB device on my system report. I tried connecting it with OS Mojave 0.13 on another system and the controller worked.

PipCave, 2019-07-08:
I can confirm that I have the same result with 0.16.11 and Alpha 3.

### Type of Controller
Replace this text Xbox 360

### OS Version
Catalina Public beta
### Driver Version
Alpha 3 and 0.16.11
### Connection Method
Wired
### Device Name and Info
N/A

### Response of Controller
Green light flashes for a couple of seconds then goes off.

### Response in Preference Pane
No device found

### Detailed Explaination of Issue
Working perfectly before installing public beta. Have gone back to Mojave and all OK so reinstalled beta and back to no devices found.

I know that both the driver and OS are in beta so do not expect a quick fix but thought you would like a report anyway.

kabutar111, 2019-07-18:
I have same issue on wireless controller

NotTheVacuum, 2019-07-18:
@FranticRain has already said in previous reports of this issue that it is not a supported OS, and he doesn't plan to address it at this stage. It's unclear whether it's not working due to defects in the beta or planned changes that will ship when Catalina is GA.

RenatoVM, 2019-10-03:
Mac OS Catalina. 10.15 Beta (19A573a)

No device found too.

Product ID:	0x028e
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.14
  Serial Number:	05F16AC
  Speed:	Up to 12 Mb/s
  Manufacturer:	Holtek
  Location ID:	0x14200000 / 1
  Current Available (mA):	500
  Extra Operating Current (mA):	0

Edited /Library/Extensions/360Controller.kext to no avail.

pharingee, 2019-10-12:
Connect via bluetooth:
https://www.polygon.com/guides/2019/9/19/20874931/connect-pair-xbox-one-controller-iphone-ipad-apple-tv-mac

raffaeleg4, 2019-10-12:
I use it with The Elder Scrolls Online and with Bluetooth connection is a mess with the scrambled buttons. Some aren't working at all (X-Y A-B). Before macOS Catalina I used to play with USB cable and every button was just fine

moanfoto, 2019-10-17:
> Connect via bluetooth:
> https://www.polygon.com/guides/2019/9/19/20874931/connect-pair-xbox-one-controller-iphone-ipad-apple-tv-mac

Sorry, this does not work for the Xbox One Classic controller, sinse it does not have bluetooth. Only the controller from Xbox One 2 and newer, works with bluetooth, not even the Elite controller :)

Regards :D

FranticRain, 2019-10-21:
Alpha 4 now works with Catalina.

closed 2019-10-21 by FranticRain