360Controller/360Controller #917: Wired controller not found on Mojave

nikhilkhanna, 2019-07-07:
### Type of Controller
Xbox 360
### OS Version
10.14.5

### Driver Version
latest

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
Device Name: Controller
  Product ID:	0x028e
  Vendor ID:	0x045e  (Microsoft Corporation)

### Response of Controller
It lights up briefly then stays dark

### Response in Preference Pane
It isn't found in preference pane

### Detailed Explaination of Issue
It isn't found in the system preferences even though my usb recognizes it as a controller

bferg88, 2019-07-08:
I posted this same bug 2 day ago #915 
Glad to see I'm not alone.

bferg88, 2019-07-08:
@nikhilkhanna since it sounds like you have the same problem as me, you might want to check how I solved it. Just need to re-open the approval window for the developer. See my comment on #915.

FranticRain, 2019-10-24:
Can you run this command and your terminal and tell me what the output is? It will prompt you for your password, but this is just because it is a tool built in to macOS that is trying to manually load the driver.
```bash
sudo kextutil /Library/Extensions/360Controller.kext
```

closed 2019-10-24 by nikhilkhanna