360Controller/360Controller #918: Support for Razer Wolverine Ultimate

kdulzo, 2019-07-07:
# Type of Controller
Razer Wolverine Ultimate

# OS Version
10.12.6

# Driver Version
0.16.8

# Connection Method
Wired

# Device Name and Info
Device Name: Razer Wolverine Ultimate
Product ID:0x0a14
Vendor ID:0x1532

# Response of Controller
Lights up and vibrates
Response in Preference Pane
Appears in preference pane but the buttons don't respond

# Detailed Explaination of Issue
New Device needs special packet support, see [#667], [#722]. [#796], [#822].
Packet capture [here](https://www.dropbox.com/s/3w86a9h05wtxg70/wolverineultimate.pcapng).

seanhealystudio, 2020-08-26:
Greetings,

I'm a Mac user (16" Macbook Pro, OS Catalina), and am trying to get my Razer Wolverine Ultimate to work. I've installed the latest version of the 'Xbox 360 Controllers' driver, which recognizes the controller but does not detect any inputs.

I must confess that I have absolutely no idea how to make any use of the packet capture that is posted above; does this file need to be dragged into a particular folder? Run in the console? Any guidance here would be very much appreciated.

Thanks,

Sean

kdulzo, 2020-08-26:
To my knowledge the device is _still_ not supported.  The packet capture is not for end user support, it's to aid developers to mapping the control sequences correctly for the device.  No one has yet.  While I was happy and capable to provide the details, I am not a developer for this project nor do I have any interest in doing so.