360Controller/360Controller #919: Hori controller (XInput compatible) no longer detected with v1.0.0-alpha.3

godly-devotion, 2019-07-12:
### Type of Controller
Hori Nintendo Switch GC Style Controller (XInput compatible)

https://www.amazon.com/Nintendo-Switch-Battle-Zelda-GameCube-Controller/dp/B01AD478L6/ref=sr_1_3

### OS Version
macOS 10.14.5

### Driver Version
360Controller 1.0.0-alpha.3 (doesn't work)
360Controller 0.16.11 (works!)

### Connection Method
Wired through USB

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: HORIPAD S
<br/>Product ID: 0x00dc
<br/>Vendor ID: 0x0f0d  (HORI CO., LTD.)

### Issue
I have a XInput compatible USB GameCube style controller made by Hori. Using driver version `0.16.11` my Mac is able to identify the controller with no issues. However when I tried to use driver version `1.0.0-alpha.3`, the controller no longer shows up. Uninstalling and reinstalling the old version makes the controller work again.

### Response in Preference Pane
No response or controllers available in Pref Panel with newer driver version.

FranticRain, 2019-10-24:
This controller is not supported by official Xbox hardware. As such, its support is outside the scope of this driver. We cannot and will not support this controller.

closed 2019-10-24 by FranticRain

jackocnr, 2020-04-25:
Just a note to say this is working for me with the latest version as of today: v1.0.0-alpha.5

With the Nintendo Switch Battle Pad (USB). Thanks!

iscaro, 2022-11-19:
Where can I find a **macOS** driver for the **Hori Battle Pad - Controller USB** now ?