360Controller/360Controller #922: Kernel panic when disconnect button clicked in system preferences.

zpchavez, 2019-07-30:
### Type of Controller
Xbox 360

### OS Version
10.14.5

### Driver Version
1.0.0 alpha 3

### Connection Method
Wireless

### Detailed Explaination of Issue
Clicking the disconnect button in system preferences will cause a pinwheel followed a few seconds later by a crash. 

Here's the first bit of the crash report. There's more but I don't think it would be helpful.

```
"caused_by":"macos","macos_version":"Mac OS X 10.14.5 (18F132)","os_version":"Bridge OS 3.5 (16P5125)","macos_system_state":"running","incident_id":"8A759F11-164D-486E-8EC1-6196B07079DA","timestamp":"2019-07-30 00:13:13.73 +0000","bug_type":"210"}
{
  "binaryImages" : [
    [
      "9a105fd5-1e36-d081-7669-0245eba3f5d1",
      18446743936270598144,
      "U"
    ],
    [
      "00000000-0000-0000-0000-000000000000",
      0,
      "A"
    ],
    [
      "a8890f6c-fbaf-3c45-ad23-8b5aadf27233",
      6442450944,
      "S"
    ],
    [
      "51868cdc-e085-3d9b-baf7-7fcc5e2eee63",
      4331274240,
      "P"
    ],
    [
      "e015f436-676a-3d5b-b02c-e622d4340ca7",
      4331552768,
      "P"
    ],
    [
      "6623ea6b-5785-35e8-beff-58084dcecdad",
      4367826944,
      "P"
    ],
    [
      "0cff77ae-8b4e-3419-8ef2-62c4765f2b3c",
      4362584064,
      "P"
    ],
    [
      "ccf9ad6d-87b2-32bf-a951-2d78aea6f23e",
      4344954880,
      "P"
    ],
    [
      "005ac577-6a2f-3188-883a-40f58440f3e7",
      4341514240,
      "P"
    ],
    [
      "f401fe9a-6b4b-3bea-8c3b-bf4fb9666d27",
      4296310784,
      "P"
[... 61 lines cut ...]
    [
      "9deec308-0827-3097-8d47-da83744db3e3",
      4336582656,
      "P"
    ],
    [
      "7b73efdf-f6e9-3edc-9308-5a4729c76aa7",
      4370300928,
      "P"
    ],
    [
      "6253e4e0-0573-3332-93aa-3918564b181b",
      4378165248,
      "P"
    ]
  ],
  "build" : "Bridge OS 3.5 (16P5125)",
  "crashReporterKey" : "c0dec0dec0dec0dec0dec0dec0dec0dec0de0001",
  "date" : "2019-07-30 00:13:13.53 +0000",
  "incident" : "8A759F11-164D-486E-8EC1-6196B07079DA",
  "kernel" : "Darwin Kernel Version 18.6.0: Thu Apr 25 21:46:37 PDT 2019; root:xnu-4903.261.4~1\/RELEASE_ARM64_T8010",
  "macOSOtherString" : "\n** In Memory Panic Stackshot Succeeded ** Bytes Traced 527040 **\n",
  "macOSPanicFlags" : "0x4",
  "macOSPanicString" : "panic(cpu 0 caller 0xffffff800302a556): \"WirelessDevice[0x100001250]::terminate(kIOServiceSynchronous) timeout\n\"@\/BuildRoot\/Library\/Caches\/com.apple.xbs\/Sources\/xnu\/xnu-4903.261.4\/iokit\/Kernel\/IOService.cpp:2428\nBacktrace (CPU 0), Frame : Return Address\n0xffffff81fdda3600 : 0xffffff80029aea2d \n0xffffff81fdda3650 : 0xffffff8002ae9e95 \n0xffffff81fdda3690 : 0xffffff8002adb70a \n0xffffff81fdda3700 : 0xffffff800295bb40 \n0xffffff81fdda3720 : 0xffffff80029ae447 \n0xffffff81fdda3840 : 0xffffff80029ae293 \n0xffffff81fdda38b0 : 0xffffff800302a556 \n0xffffff81fdda3910 : 0xffffff800302e07a \n0xffffff81fdda3990 : 0xffffff7f868fd319 \n0xffffff81fdda3e00 : 0xffffff7f868fcf7c \n0xffffff81fdda3e40 : 0xffffff7f833ea7e8 \n0xffffff81fdda3e90 : 0xffffff7f833a1d55 \n0xffffff81fdda3ef0 : 0xffffff7f833a1c60 \n0xffffff81fdda3f30 : 0xffffff8003056362 \n0xffffff81fdda3f70 : 0xffffff80030558bc \n0xffffff81fdda3fa0 : 0xffffff800295b0ce \n      Kernel Extensions in backtrace:\n         com.apple.driver.usb.AppleUSBCommon(1.0)[4505E593-FEDF-3B1F-B54C-452AE387CB88]@0xffffff7f833a0000->0xffffff7f833a7fff\n         com.apple.iokit.IOUSBHostFamily(1.2)[328248B8-6904-34F4-9BC4-7F94CD7EE484]@0xffffff7f833a8000->0xffffff7f83421fff\n            dependency: com.apple.driver.AppleBusPowerController(1.0)[FD6B2521-541A-3BFF-A921-502B058BD3CC]@0xffffff7f83398000\n            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[4505E593-FEDF-3B1F-B54C-452AE387CB88]@0xffffff7f833a0000\n         com.mice.driver.WirelessGamingReceiver(1.0)[7527B808-44B0-3E62-BC88-6263B44751C3]@0xffffff7f868fb000->0xffffff7f86905fff\n            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[43464787-35A5-36AA-92F4-D76B57A1B411]@0xffffff7f83248000\n            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[328248B8-6904-34F4-9BC4-7F94CD7EE484]@0xffffff7f833a8000\n\nBSD process name corresponding to current thread: kernel_task\n\nMac OS version:\n18F132\n\nKernel version:\nDarwin Kernel Version 18.6.0: Thu Apr 25 23:16:27 PDT 2019; root:xnu-4903.261.4~2\/RELEASE_X86_64\nKernel UUID: 7C8BB636-E593-3CE4-8528-9BD24A688851\nKernel slide:     0x0000000002600000\nKernel text base: 0xffffff8002800000\n__HIB  text base: 0xffffff8002700000\nSystem model name: Macmini8,1 (Mac-7BA5B2DFE22DDD8C)\n\nSystem uptime in nanoseconds: 350919822271285\nlast loaded kext at 73134996422231: com.mice.driver.Wireless360!C\t1.0 (addr 0xffffff7f864b8000, size 20480)\nlast unloaded kext at 9264582007926: com.mice.driver.Wireless360!C\t1.0 (addr 0xffffff7f864b8000, size 12288)\nloaded kexts:\ncom.mice.driver.Wireless360!C\t1.0\ncom.Cycling74.driver.Soundflower\t2\ncom.mice.driver.WirelessGamingReceiver\t1.0\n>!A!BMultitouch\t96\n>AudioAUUC\t1.70\n>!AGraphicsDevicePolicy\t3.40.16\n@fileutil\t20.036.15\n@AGDCPluginDisplayMetrics\t3.40.16\n>!AHV\t1\n|IOUserEthernet\t1.0.1\n|IO!BSerialManager\t6.0.12f1\n>!AUpstreamUserClient\t3.6.5\n>pmtelemetry\t1\n>!AMCCSControl\t1.5.9\n@Dont_Steal_Mac_OS_X\t7.0.0\n>AGPM\t110.25.10\n>!APlatformEnabler\t2.7.0d0\n>X86PlatformShim\t1.0.0\n>!A!IKBLGraphics\t12.0.9\n>!AVirtIO\t2.1.3\n>BridgeAudioCommunication\t5.46\n>!AThunderboltIP\t3.1.2\n>!ABridgeAudio!C\t5.46\n>!AGFXHDA\t100.1.414\n>!A!ICFLGraphicsFramebuffer\t12.0.9\n>!A!ISlowAdaptiveClocking\t4.0.0\n>!AOSXWatchdog\t1\n>!A!IPCHPMC\t2.0.1\n>!AAVEBridge\t6.18\n@filesystems.autofs\t3.0\n>BCMWLANFirmware4355.Hashstore\t1\n>BCMWLANFirmware4364.Hashstore\t1\n>BCMWLANFirmware4377.Hashstore\t1\n|!ABCM5701Ethernet\t10.3.3\n>!ASDXC\t1.7.6\n>!ABCMWLANBusInterfacePCIe\t1\n@filesystems.hfs.kext\t407.200.4\n@BootCache\t40\n@!AFSCompression.!AFSCompressionTypeDataless\t1.0.0d1\n@!AFSCompression.!AFSCompressionTypeZlib\t1.0.0\n@!ASystemPolicy\t1.0\n@filesystems.apfs\t945.260.7\n@private.KextAudit\t1.0\n>!AACPIButtons\t6.1\n>!ASMBIOS\t2.1\n>!AACPIEC\t6.1\n>!AAPIC\t1.7\n@nke.applicationfirewall\t201\n$TMSafetyNet\t8\n>!AMultitouchDriver\t2450.1\n>!AInputDeviceSupport\t2440.2\n>!AGraphicsControl\t3.40.16\n|IOAVB!F\t740.1\n@plugin.IOgPTPPlugin\t740.2\n>!ASSE\t1.0\n>!ASMBus!C\t1.0.18d1\n>!A!BHIDKeyboard\t208\n>!AHIDKeyboard\t208\n>IO!BHIDDriver\t6.0.12f1\n@!AGPUWrangler\t3.40.16\n|IOAudio!F\t206.5\n@vecLib.kext\t1.2.0\n|IONDRVSupport\t530.51\n|IO!BHost!CUARTTransport\t6.0.12f1\n|IO!BHost!CTransport\t6.0.12f1\n>!A!ILpssUARTv1\t3.0.60\n>!A!ILpssUARTCommon\t3.0.60\n>!AOnboardSerial\t1.0\n|IOSkywalk!F\t1\n@!AGraphicsDeviceControl\t3.40.16\n|IOAccelerator!F2\t404.9\n|IOGraphics!F\t530.66\n|IOSurface\t255.5.4\n|IOSlowAdaptiveClocking!F\t1.0.0\n>X86PlatformPlugin\t1.0.0\n>IOPlatformPlugin!F\t6.0.0d8\n@kext.triggers\t1.0\n>usb.IOUSBHostHIDDevice\t1.2\n>usb.cdc.ncm\t5.0.0\n>usb.cdc\t5.0.0\n>usb.networking\t5.0.0\n>usb.!UHostCompositeDevice\t1.2\n>!UHostMergeProperties\t1.2\n|IOEthernetAVB!C\t1.1.0\n>!ABCMWLANCore\t1.0.0\n>mDNSOffloadUserClient\t1.0.1b8\n>IOImageLoader\t1.0.0\n|IOSerial!F\t11\n|IO80211!FV2\t1200.12.2\n>corecapture\t1.0.4\n>!AThunderboltPCIDownAdapter\t2.1.4\n>!AThunderboltDPInAdapter\t5.6.4\n>!AThunderboltDPAdapter!F\t5.6.4\n>!AHPM\t3.4.1\n>!A!ILpssI2C!C\t3.0.60\n>!A!ILpssDmac\t3.0.60\n>!A!ILpssI2C\t3.0.60\n@filesystems.hfs.encodings.kext\t1\n>!AThunderboltNHI\t4.7.6\n|IOThunderbolt!F\t6.8.4\n>usb.!UVHCIBCE\t1.2\n>usb.!UVHCI\t1.2\n>usb.!UVHCICommonBCE\t1.0\n>usb.!UVHCICommon\t1.0\n>!AEffaceableNOR\t1.0\n|IOBufferCopy!C\t1.1.0\n|IOBufferCopyEngine!F\t1\n|IONVMe!F\t2.1.0\n>usb.!UHostPacketFilter\t1.0\n|IOUSB!F\t900.4.2\n>usb.!UXHCIPCI\t1.2\n>usb.!UXHCI\t1.2\n>!AEFINVRAM\t2.1\n>!AEFIRuntime\t2.1\n>!ASMCRTC\t1.0\n|IOSMBus!F\t1.1\n|IOHID!F\t2.0.0\n$quarantine\t3\n$sandbox\t300.0\n@kext.!AMatch\t1.0.0d1\n>!AKeyStore\t2\n>!UTDM\t456.260.3\n>!AMobileFileIntegrity\t1.0.5\n|IOUSBMass!SDriver\t145.200.2\n|IOSCSIBlockCommandsDevice\t408.250.3\n|IOSCSIArchitectureModel!F\t408.250.3\n@kext.CoreTrust\t1\n>!ACredentialManager\t1.0\n>KernelRelayHost\t1\n>!ASEPManager\t1.0.1\n>IOSlaveProcessor\t1\n>!AFDEKeyStore\t28.30\n>!AEffaceable!S\t1.0\n|IOTimeSync!F\t740.2\n|IONetworking!F\t3.4\n>DiskImages\t493.0.0\n|IO!S!F\t2.1\n|IO!B!F\t6.0.12f1\n|IOUSBHost!F\t1.2\n>usb.!UCommon\t1.0\n>!ABusPower!C\t1.0\n|IOReport!F\t47\n>!AACPIPlatform\t6.1\n>!ASMC\t3.1.9\n|IOPCI!F\t2.9\n|IOACPI!F\t1.4\n@kec.Libm\t1\n@kec.pthread\t1\n@kec.corecrypto\t1.0\n",
```

zpchavez, 2019-07-30:
Another issue which I suspect is related is the game freezing when it attempts to stop controller rumble. It seems to happen at random, but also when a controller disconnects and it tries to stop rumble in the Dispose method. Here are two examples of those thread dumps. I can make this a separate issue if I'm incorrect about them being related.

```
\Full thread dump:

"<threadpool thread>" tid=0x0x7000030e4000 this=0x0x11ce73000 thread handle 0x11e state : interrupted state owns ()

"IO Threadpool worker" tid=0x0x700003127000 this=0x0x11ce4aea0 thread handle 0x11f state : interrupted state owns ()

"Threadpool worker" tid=0x0x70000398e000 this=0x0x11cd85618 thread handle 0x10b state : interrupted state owns ()

"Threapool monitor" tid=0x0x70000378b000 this=0x0x11cd85750 thread handle 0x10a state : interrupted state owns ()

"<unnamed thread>" tid=0x0x119c225c0 this=0x0x10ffe0ea0 thread handle 0x103 state : not waiting owns ()
  at (wrapper managed-to-native) SDL2.SDL.SDL_HapticStopAll (intptr) <0xffffffff>
  at Microsoft.Xna.Framework.SDL2_FNAPlatform.SetGamePadVibration (int,single,single) <0x000cb>
  at (wrapper delegate-invoke) <Module>.invoke_bool__this___int_single_single (int,single,single) <0xffffffff>
  at Microsoft.Xna.Framework.Input.GamePad.SetVibration (Microsoft.Xna.Framework.PlayerIndex,single,single) <0x0004f>
  at Monocle.MInput/XGamepadData.StopRumble () <0x00055>
  at Monocle.MInput/XGamepadData.Update (bool) <0x0012d>
  at Monocle.MInput.Update () <0x00126>
  at Monocle.Engine.Update (Microsoft.Xna.Framework.GameTime) <0x00305>
  at TowerFall.TFGame.Update (Microsoft.Xna.Framework.GameTime) <0x0002f>
  at Microsoft.Xna.Framework.Game.Tick () <0x00579>
  at Microsoft.Xna.Framework.SDL2_FNAPlatform.RunLoop (Microsoft.Xna.Framework.Game) <0x00dbe>
  at Microsoft.Xna.Framework.Game.Run () <0x00077>
  at TowerFall.TFGame.Main (string[]) <0x0049e>
  at (wrapper runtime-invoke) <Module>.runtime_invoke_void_object (object,intptr,intptr,intptr) <0xffffffff>
```

```
^\Full thread dump:

"IO Threadpool worker" tid=0x0x700008855000 this=0x0x116ad94e0 thread handle 0x11f state : interrupted state owns ()

"Threapool monitor" tid=0x0x700008eb9000 this=0x0x116438750 thread handle 0x10a state : interrupted state owns ()

"Threadpool worker" tid=0x0x7000090bc000 this=0x0x116438618 thread handle 0x10b state : interrupted state owns ()

"<unnamed thread>" tid=0x0x1162d65c0 this=0x0x109f84ea0 thread handle 0x103 state : not waiting owns ()
  at (wrapper managed-to-native) SDL2.SDL.SDL_HapticClose (intptr) <0xffffffff>
  at Microsoft.Xna.Framework.SDL2_FNAPlatform.INTERNAL_RemoveInstance (int) <0x000d5>
  at Microsoft.Xna.Framework.SDL2_FNAPlatform.RunLoop (Microsoft.Xna.Framework.Game) <0x00b6c>
  at Microsoft.Xna.Framework.Game.Run () <0x00077>
  at TowerFall.TFGame.Main (string[]) <0x0049e>
  at (wrapper runtime-invoke) <Module>.runtime_invoke_void_object (object,intptr,intptr,intptr) <0xffffffff>

"<threadpool thread>" tid=0x0x700008812000 this=0x0x116ad9618 thread handle 0x11e state : interrupted state owns ()
```

Tursko, 2019-09-03:
For whatever reason, if I turn off the driver in the advanced setting the controller still works and the kernel panic doesn't happen anymore.

Xbox 360 Wireless Adapter