360Controller/360Controller #926: Wired XBox360 controller not showing up in new Alpha

CSho27, 2019-08-31:
### Type of Controller
XBox360 Wired Guitar Hero Guitar (X-plorer)

### OS Version
High Sierra 10.13.6

### Driver Version
360Controller 1.0.0-alpha.3

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Guitar Hero X-plorer
<br/>Product ID: 0x4748
<br/>Vendor ID: 0x1430

### Response of Controller
It occasionally blinks and even seemed to register as player 1 for a moment but then shut back off

### Response in Preference Pane
When I go to system preferences, it says no devices could be found. It just isn't registering at all.

### Detailed Explaination of Issue
I downloaded the newest alpha and installed it, but when I went to system preferences to run the controller tester it could not find the device. There must be some sort of regression, because I downloaded the last release (360Controller 0.16.11) and it worked just fine. I wish I could give you a more detailed explanation, but I basically tried the alpha and nothing showed up, then tried the latest release and it worked perfectly.

PokeCaptain, 2019-09-01:
As per [this](https://github.com/360Controller/360Controller/issues/156#issuecomment-524668398) comment, 360Controller is no longer being actively maintained and this issue will go unresolved. Follow #156 for a possible future replacement.

aandroyd, 2019-10-07:
OS 10.13.6, same issue. I regressed from 1.0.0alpha3 to 0.16.6 and now it works!

FranticRain, 2019-10-24:
Can you run this command in your terminal and tell me what the output is? It will ask you for your password, but this is just because it's a macOS tool that is attempting to manually load the driver.
```
sudo kextutil /Library/Extensions/360Controller.kext
```

Eweie, 2019-10-25:
> Can you run this command in your terminal and tell me what the output is? It will ask you for your password, but this is just because it's a macOS tool that is attempting to manually load the driver.
> 
> ```
> sudo kextutil /Library/Extensions/360Controller.kext
> ```
I have the same issue. here is my output using alpha 4 but it was the same in 3.

Notice: /Library/Extensions/360Controller.kext has debug properties set.
Can't open CFBundle for /System/Library/Extensions/roller.kext.
Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 0.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 0.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV9IOService' for vtable '__ZTV17Xbox360Peripheral' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22Xbox360Pretend360Class was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOneControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOnePretend360Class was not patched because its parent, the __ZTV22XboxOneControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV27XboxOriginalControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) Can't load kext com.mice.driver.Xbox360Controller - link failed.
(kernel) Failed to load executable for kext com.mice.driver.Xbox360Controller.
(kernel) Kext com.mice.driver.Xbox360Controller failed to load (0xdc008016).
(kernel) Failed to load kext com.mice.driver.Xbox360Controller (error 0xdc008016).
Failed to load /Library/Extensions/360Controller.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).

FranticRain, 2019-10-25:
@Eweie I’ve got a pretty good idea of what the issue is from that. I was already working on a fix for this in order to get better Catalina support. Give me some time and I’ll put out an alpha 5 that should fix this.