360Controller/360Controller #928: doesn't let me uninstall.

IcyMelonn, 2019-09-04:
### Type of Controller
N/A

### OS Version
Mojave 10.14.5
### Driver Version
1.0.0

### Connection Method
N/A

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
N/A

### Detailed Explaination of Issue
I want to uninstall the driver however when I click uninstall it gives me an error everytime.

greystitch, 2019-09-23:
After the password prompt, it shows the window with error "Cannot uninstall". However, the kexts are being deleted.

JohnCWakley, 2019-10-15:
I am having the same issue on Catalina 10.15. I installed 360ControllerInstall_1.0.0-alpha.3.dmg.

Any help would be appreciated.

FranticRain, 2019-11-04:
Hey everybody who is having trouble uninstalling. The alpha 5 release now includes a script called `uninstall.command`. If you run this script (double click on it) on your machine, it should uninstall all of the products of 360Controller. Let me know if it works for you.

Tina-King, 2021-06-04:
It doesnt work for me on Big Sur 11.2.3

I had to `chmod a+x uninstall.command` and after running it shows it completed uninstalling but I still see the software installed on my system preferences. I would like to clean it up, any ideas?

Thanks

idhammi, 2021-06-29:
> It doesnt work for me on Big Sur 11.2.3
> 
> I had to `chmod a+x uninstall.command` and after running it shows it completed uninstalling but I still see the software installed on my system preferences. I would like to clean it up, any ideas?
> 
> Thanks

Have same issues, you can right click on the 360controller on the system preferences and select remove