360Controller/360Controller #929: 360 wired controller not working high Sierra

poomperzuhhh, 2019-09-05:
### Type of Controller
Xbox 360, Black controller. Original controller with wireless capability but hooked up to Mac using original charging cable. Battery removed from controller.

### OS Version
High Sierra, 10.13.6

### Driver Version
V1.0.0 Alpha.3

### Connection Method
Wired with original charging cable

### Device Name and Info

Device Name: "Vendor specific device" - No device name coming up
Product ID: 0x028f
Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Controller lights up with continuous flashing at regular pace. No vibration.

### Response in Preference Pane
Controller does not appear in preference pane

### Detailed Explaination of Issue
After installing the latest xbox 360 controller driver, I have connected my controller to my Mac (high Sierra, 10.13.6) using the charging cable that comes with the standard rechargeable 360 battery. I have gamed on the actual 360 using this cable before when I haven't had a battery present, so I am presuming the cable can transmit data to and from the device it is connected to and doesn't solely just act as a means of providing power to it.

Xbox 360 controller appears In the systems preferences, but when I click on it, it doesn't recognise my controller at any point.

Any solutions?

Thanks! :)

valkyriesavage, 2019-09-07:
Same, but I'm on Mojave (10.14.6). :( I saw that someone else mentioned that this whole project is being deprecated (see #156 ), so not clear if we'll get an answer.

FranticRain, 2019-09-07:
Xbox 360 wireless controllers are always wireless controllers. They don’t turn into wired controllers when you plug in the charging cable. This isn’t an issue with the driver and is documented in the README.

closed 2019-09-07 by FranticRain