360Controller/360Controller #931: When I plug in my Xbox One controller it vibrates but doesn't actually detect it.

BladeSlice17, 2019-09-09:
### Type of Controller
Xbox One

### OS Version
10.14.6

### Driver Version
0.16.11

### Connection Method
Wired

### Device Name and Info
N/A

### Response of Controller
Vibrates but does not light up

### Response in Preference Pane
Says controller not found in Preference Plane

### Detailed Explaination of Issue
When I plug in my Xbox One controller it vibrates and nothing else. I try to turn it on but it acts like its looking for an Xbox with the guide button blinking

Moyal0621, 2019-09-25:
I have the same issue.

L0VEFIRE, 2019-10-09:
BladeSlice, did you ever attempt to connect it via bluetooth prior to it not working?? curious? also usually doing a MacOS reinstall will resolve most driver issue in my experiences

leohsmedeiros, 2019-10-23:
I am having the same issue

FranticRain, 2019-10-23:
You must update your original post with your controller information. A guide on how to do so can be found here: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id.

Controller information is only not required for issue that do not involve controllers. For example, a feature request that isn't specific to any controller, or maybe a typo somewhere in the code or documentation. For all issues individuals have with getting a controller working, controller information is ABSOLUTELY VITAL to solving your problem. Not only does it let us know if your controller is already supported, it also lets us know that the operating system can see the controller and is a crucial step in resolving your issue.

Since this has turned into a dogpile issue, I am going to ask that all of you open individual issues that follow the template, as I imagine that each of you has a different setup and therefore a different problem.

closed 2019-10-23 by FranticRain