360Controller/360Controller #932: Hyperkin Duke not detected by driver

mex20, 2019-09-11:
### Type of Controller
Xbox One; Hyperkin Duke controller

### OS Version
10.13.6

### Driver Version
1.0.0-alpha3

### Connection Method
wired through microusb.

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Hyperkin Duke Controller
<br/>Product ID: 0x0652
<br/>Vendor ID: 0x2e24

### Response of Controller
Controller vibrates upon wired connection initiation, for a second or two and then is not detected by the driver.

### Response in Preference Pane
The preference pane does not react to controller movement and displays "no devices found".

### Detailed Explaination of Issue
The driver does not detect the controller when it is attached. The controller is viewable in the system information window, under the USB host controller and named "Hyperkin Duke" with the above product ID and vendor ID listed above. The controller will vibrate when plugged in, indicating that power is being received by the controller. I am using the provided by Hyperkin USB cable. The controller works my PC and Xbox One.

rullinoiz, 2019-10-22:
hey, im having the same issue. does the boot up thing on the xbox button show when you plug it in? mine doesnt

FranticRain, 2019-11-12:
Here's a build with this controller added. Let me know if you have any issues with it. All future builds will include this controller.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3834384/360ControllerInstall.dmg.zip)

closed 2019-11-12 by FranticRain