360Controller/360Controller #933: Mac won't start when drivers are installed on 10.14.6

ctoothaker1, 2019-09-26:
### Type of Controller
N/A
### OS Version
macOS Mojave 10.14.6
### Driver Version
1.0.0 Alpha 3
### Connection Method
N/A
### Device Name and Info
N/A
### Response of Controller
N/A
### Response in Preference Pane
N/A
### Detailed Explaination of Issue
I installed the 360ControllerInstall_1.0.0-alpha.3.dmg and the install went normally. I then restarted my mac and after I entered my password, the computer froze permanently. After some googling and learning how to use terminal, I was able to move the .kext files that were installed into a new folder through terminal when I started up my mac when holding Command+S on the keyboard. Getting to the point, I believe the drivers are not supported on my version of macOS since while they were installed, my computer would not start up. My mac is a Mid 2014 Retina 13 inch MacBook Pro. During this whole thing, my controller wasn't even plugged in.

Jukkauuno, 2019-09-27:
Same thing happened here as well.

Would you be able to give some help on solving the issue in Terminal?..

ctoothaker1, 2019-09-30:
> Same thing happened here as well.
> 
> Would you be able to give some help on solving the issue in Terminal?..

https://apple.stackexchange.com/questions/367488/loading-bar-stuck-at-the-very-end-after-entering-login-credentials

The forum response on the bottom of that webpage helped me solve the problem. When you turn on your mac, hold Command + S and then follow the steps.