360Controller/360Controller #934: 360 Controller not recognised

ForkFaceJones, 2019-10-01:
### Type of Controller
Xbox360

### OS Version
10.14.6 (18G103)

### Driver Version
Version 0.16.11 - I have tried 2, 3 and a few of the ones you have linked here. And the Notarised. 

### Connection Method
Wired

### Device Name and Info
>Device Name: Vendor Specific Device
<br/>Product ID: 0x028f
<br/>Vendor ID: 0x045e  (Microsoft Corporation)
  Version:	1.06
  Location ID:	0x14100000

### Response of Controller
Guide button lights nothing else. On a previous install of the driver (2 & 3) the comp recognises the charge cable has been plugged in. But not the controller.

### Response in Preference Pane
No device

### Detailed Explaination of Issue
Response from
<br/> sudo kextutil /Library/Extensions/360Controller.kext

Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

I am wondering if the folder is restricted requiring a boot into safe mode or something to turn off system protection and change the restrictions on the kext folder -
ls -laO /Library/StagedExtensions/
total 0
drwxr-xr-x@  4 root  wheel  restricted  128 30 Sep 22:14 .
drwxr-xr-x+ 67 root  wheel  sunlnk     2144 28 Jul 18:58 ..
drwxr-xr-x   5 root  wheel  restricted  160 30 Sep 22:48 Library
drwxr-xr-x   3 root  wheel  restricted   96 30 Sep 22:14 System

I can get the device partially recognised in VM.

And it shows up here in system profile under USB
Vendor-Specific Device:
  Product ID:	0x028f
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.06
  Location ID:	0x14100000

<br/>Console references:
2019-10-01 01:41:57.109945 +0100	default	01:41:57.109945 +0100	kernel	Xbox360Peripheral::probe(IOUSBHostDevice)
2019-10-01 01:41:57.109952 +0100	default	01:41:57.109952 +0100	kernel	Xbox360Peripheral::start(IOUSBHostDevice) <1>
2019-10-01 02:00:40.496181 +0100	default	02:00:40.496181 +0100	kernel	Xbox360Peripheral::probe(IOUSBHostDevice)
2019-10-01 02:00:40.496190 +0100	default	02:00:40.496190 +0100	kernel	Xbox360Peripheral::start(IOUSBHostDevice) <1>
2019-10-01 02:00:56.742654 +0100	default	02:00:56.742654 +0100	kernel	Xbox360Peripheral::probe(IOUSBHostDevice)
2019-10-01 02:00:56.742666 +0100	default	02:00:56.742666 +0100	kernel	Xbox360Peripheral::start(IOUSBHostDevice) <1>
2019-10-01 02:21:02.471549 +0100	default	02:21:02.471549 +0100	kernel	Xbox360Peripheral::probe(IOUSBHostDevice)
2019-10-01 02:21:02.471561 +0100	default	02:21:02.471561 +0100	kernel	Xbox360Peripheral::start(IOUSBHostDevice) <1>
2019-10-01 02:21:04.987113 +0100	default	02:21:04.987113 +0100	kernel	Xbox360Peripheral::probe(IOUSBHostDevice)
2019-10-01 02:21:04.987121 +0100	default	02:21:04.987121 +0100	kernel	Xbox360Peripheral::start(IOUSBHostDevice) <1>

I think that's all the information I have. Charge cable is original I think. I have four pads I have tried, none register.

Thanks for any help :)

artfail, 2019-10-03:
I think I have the same issue. Driver worked during install. After install and reboot, my wired xbox360 controller is no longer detected.

FranticRain, 2019-10-04:
For Xbox 360 controllers, wireless controllers don't become wired controllers by connecting a play and charge kit. You'll need to use a wireless dongle to communicate with one. As for the issue, I think that might be related to the fact that my developer certificate is expired. Someone else is going to have to start signing the driver. 🤷‍♂

FranticRain, 2019-10-24:
Closing due to the fact that using a play and charge kit to use a controller isn't possible.

closed 2019-10-24 by FranticRain