360Controller/360Controller #936: not working on mac os Catalina version 10.15. but used to work perfectly before the update.

that-weird-coder, 2019-10-09:
### Type of Controller
Xbox One
### OS Version
version 10.15
### Driver Version
360Controller 1.0.0-alpha.3
### Connection Method
usb cable
### Device Name and Info
MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports)
### Response of Controller
it vibrates then nothing happens. it doesn't show in the connected devices.

### Response in Preference Pane
it doesn't appear in the pane.

### Detailed Explaination of Issue
i updated my my os then connected my controller to play some games it vibrated but the white light on the xbox logo in the controller didn't light up. i opend the preference pane and the controller wasn't connected.

bob12345q, 2019-10-09:
same also doesn't work with the Xbox 360 controller and the wireless receiver.

MontyTheMooch, 2019-10-09:
Same issue. Worked prior to update. Unable to tell if it's because 360Controller is 32bit. Catalina is 64bit only.

MyBadOops, 2019-10-09:
Yeh, same here, really missing this.

L0VEFIRE, 2019-10-09:
I was using driver 0.16.10 via wired connection just fine with Catalina (public release), when you attempt to use wireless it breaks it.

trankkilou, 2019-10-09:
> I was using driver 0.16.10 via wired connection just fine with Catalina (public release), when you attempt to use wireless it breaks it.

Can you tell me where did you find this driver? I have a wired 360 and it doesn't work with the last alpha version

L0VEFIRE, 2019-10-09:
its simple as searching google for "xbox driver Mac" one of the first links for further reference. that says "Releases · 360Controller/360Controller · GitHub"

here is the link: 
https://github.com/360Controller/360Controller/releases

scroll down till you find 0.16.10. (if that doesn't work, let me know)

trankkilou, 2019-10-09:
Thx ! I was using this page but I didn’t scroll down enough to see the stable release :D

callum504, 2019-10-09:
can confirm, yeah that does not work unfortunately for me too

L0VEFIRE, 2019-10-10:
The only fix I know should work is uninstalling the xbox controller driver and then doing a macOS reinstall in OS Recovery, to reinstall all the OS files, (you data remains intact) Then reinstall the driver and it should work again via wired connection only!!

callum504, 2019-10-10:
will do that now! ill let you know if it works

trankkilou, 2019-10-10:
I have installed catalina yesterday. I never tried wireless as I dont have one. I have already tried to reinstalled the driver and it changes nothing

Julien

L0VEFIRE, 2019-10-10:
hmmm. what controller you have?

brettskiii, 2019-10-10:
It doesn't work after Catalina, I'm on wired mode was working fine before. Re-installed driver no luck

L0VEFIRE, 2019-10-10:
i tested this myself after installing catalina on a separate partition l, it doesn’t work wired. i believe part of the reason is catalina’s new file system separates system files from users and the xbox driver includes both.

L0VEFIRE, 2019-10-10:
I just heard back from a game company and apple, and both have said the same thing about the future of macOS and Catalina. macOS Catalina has quote "... introduced additional system security measures which can interfere with 3rd party drivers, not to mention other low level changes that may break existing drivers....The reality is Catalina is going to be a mess for a while and if you hadn’t upgraded already I would tell you to hold off for a while until things get settled; if they can be....As for apple, there sole intentions are to slow eliminate the ability to use third-party anything making everything only accessible within their own ecosystem; this will create havoc but its coming faster then we like to accept." 

So in summary, stay far far away from Catalina (that island is not for us)

leoma95, 2019-10-11:
Same here was working fine before the upgrade to Catalina. I have an Xbox One 1708 controller and an XboxOne Elite. Neither works.

I tried uninstalling and installing that 0.16.11 Notarized release.

The driver installs fine, but never detects the controller "No Devices Found"

With the XboxOne 1708 controller, I'm able to connect natively with Bluetooth and the controller works fine with Apple Arcade Games via Bluetooth. However, when I try to use it with Fortnite the key mappings are all wrong (A is B, X is Y, most of the rest is messed up or unmapped). That is why I need the USB wired driver.

L0VEFIRE, 2019-10-11:
Leoma95,

tbh, the best thing todo given my previous post above is to send feedback to Apple requesting they support wired connections for Xbox controller. bc soon 3rd party drivers wont be allowed on apples OS come next year; it what i've been hearing. Go after apple, FranticRains abilitys are limited with catalina at the moment and is unknown if it will be supported

ImShalev, 2019-10-12:
Same, can't get it to work on Catalina.

raffaeleg4, 2019-10-12:
I have the same with macOS Catalina, driver not recognised but the controller is connected by Bluetooth and works with some games altough the keys are scrambles (e.g. The Elder Scrolls Online). I used to play the game with the same controller connected by cable and always worked in macOS Mojave

leoma95, 2019-10-12:
> Leoma95,
> 
> tbh, the best thing todo given my previous post above is to send feedback to Apple requesting they support wired connections for Xbox controller. bc soon 3rd party drivers wont be allowed on apples OS come next year; it what i've been hearing. Go after apple, FranticRains abilitys are limited with catalina at the moment and is unknown if it will be supported
> 
> also apple arcade works with Mojave as well, back up, erase your partition, and downgrade to Mojave is your best options.

Well, rollback to Mojave is no longer an option for me. It would have been nice if there was notice that this was not working with Catalina beta and was going to be broken irreparably.

In case of wireless connection, all that is missing is a key mapping application.

In case of USB connection, it seems like device detection needs a tweak.

L0VEFIRE, 2019-10-12:
to everyone, here is a real easy solution. write tcook@apple.com a “nice” email about how he needs to include wired controller compatibility for their computers; since they have open noted there taken responsibility for xbox and ps4 controllers now. it your best option. since this any getting fix this year most likely

leoma95, 2019-10-13:
> to everyone, here is a real easy solution. write [tcook@apple.com](mailto:tcook@apple.com) a “nice” email about how he needs to include wired controller compatibility for their computers; since they have open noted there taken responsibility for xbox and ps4 controllers now. it your best option. since this any getting fix this year most likely

LOVEFIRE, do you have any idea why the wireless/Bluetooth connection has the keymappings messed up in Fortnite and other games? The real solution would be to use Wireless access, since that is supported natively. Any ideas on how to fix the key mappings?

L0VEFIRE, 2019-10-13:
Well, as far as I know, there is no way to adjust the keybindings for wireless since you need some sort of software to manage that aspect. unfortunately

L0VEFIRE, 2019-10-13:
well xbox is only supported natively for apple arcade since all developer for those games were made to work with xbox vis bluetooth on mac. but just using the xbox controller wireless wouldn’t work for other games bc the developers would have to support it which most don’t, and doing so isnt as easy as you’d think

FranticRain, 2019-10-19:
Alpha 4 now works with Catalina.

closed 2019-10-19 by FranticRain

MontyTheMooch, 2019-10-19:
Now works WIRED on Catalina. Wireless XB1 Controller still cannot properly map controls

FranticRain, 2019-10-19:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

This is also true of the wireless xbox one controller support.