360Controller/360Controller #940: Drivers never ask for Permission

L0VEFIRE, 2019-10-10:
### Type of Controller
Xbox One

### OS Version
Any Version of High Sierra or Mojave 

### Driver Version
all drivers from 0.16.10 to Alpha 3

### Connection Method
Wired

### Device Name and Info
iBridge:

  Product ID:	0x8600
  Vendor ID:	0x05ac (Apple Inc.)
  Version:	1.01
  Manufacturer:	Apple Inc.
  Location ID:	0x14200000

Controller:

  Product ID:	0x02ea
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	3.01
  Serial Number:	3033363030353037323632383239
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft
  Location ID:	0x14100000 / 2
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
Works Fine

### Response in Preference Pane
Works Fine

### Detailed Explanation of Issue
All your driver always say to accept permission in "Security & Privacy" Preference pane but not once has it ever asked me to do so in all versions of macOS I have used, and all the times I have had to wipe my drives and reinstall macOS. 

https://streamable.com/hqdfb

ImShalev, 2019-10-13:
I think this is why it does not work in macOS Catalina.

L0VEFIRE, 2019-10-13:
Maybe? but then consider if it never asked on mojave and always worked for me why would that suddenly change now?

L0VEFIRE, 2019-10-13:
i feel it a bit more than just that, Catalina requires more types of security parameters when using anything that’s none apple and  unidentified, or unsigned. and i don’t think the drivers have meet them. what those parameters are idk 😐

FranticRain, 2019-10-24:
@L0VEFIRE Is this issue still valid?

tonylegrone, 2019-11-01:
I resolved this for myself in Alpha 4 by adding `/Library/Application Support/MICE/360Daemon.app` to the "input Monitoring" list in the Catalina security preferences.

I have a wireless xbox one controller connected with USB and it's working now.

kstevens14, 2019-11-14:
> I resolved this for myself in Alpha 4 by adding `/Library/Application Support/MICE/360Daemon.app` to the "input Monitoring" list in the Catalina security preferences.
> 
> I have a wireless xbox one controller connected with USB and it's working now.

how do you get to Catalina security preferences to add this

kstevens14, 2019-11-14:
> I resolved this for myself in Alpha 4 by adding `/Library/Application Support/MICE/360Daemon.app` to the "input Monitoring" list in the Catalina security preferences.
> 
> I have a wireless xbox one controller connected with USB and it's working now.

how did you add this to the input monitoring?

tonylegrone, 2019-11-14:
> how did you add this to the input monitoring?

[image]

kstevens14, 2019-11-15:
> > how did you add this to the input monitoring?
> 
> [image]

its not giving me the option to add anything new to the input monitoring

FranticRain, 2019-11-15:
Can confirm.

[image]

tonylegrone, 2019-11-15:
That's super weird. Sounds like a Catalina bug. [See this post.](https://github.com/ckb-next/ckb-next/issues/431#issuecomment-536195726) You might be able to try and drag the app from Finder into the preference window. If you're forced to install something else that requests the permission (as the other user described), I know that OpenEmu requests it for joystick support.

FranticRain, 2019-11-15:
But the real, actual fix to this issue is to check the "Enable Driver" checkbox under the "Advanced" menu in the preference pane. Do that, and it will probably resolve the need to do this.

closed 2019-11-15 by FranticRain

tonylegrone, 2019-11-15:
> But the real, actual fix to this issue is to check the "Enable Driver" checkbox under the "Advanced" menu in the preference pane. Do that, and it will probably resolve the need to do this.

Are you sure that works in Catalina? When I tried that, nothing happened. I didn't get a request for the input monitoring permission and my controller still didn't work.

kstevens14, 2019-11-16:
> Are you sure that works in Catalina? When I tried that, nothing happened. I didn't get a request for the input monitoring permission and my controller still didn't work.

same thing happened to me. ive tried 4 different versions of 360Controller and none of them have worked. I have a MacBook Pro running Catalina. not sure how to get this to work. my controller is never identified

FranticRain, 2019-11-16:
Run this command in your terminal and let me know what it tells you. It will ask for your password because it’s the OS trying to load the kext.

```
sudo kextutil /Library/Extensions/360Controller.kext
```

The reason I say to check the checkbox is because it includes debugging steps to make sure that the driver is correctly loaded when you check the checkbox. If the checkbox can be checked without feedback, then the driver was loaded without a problem. This is a manual check for the same action. If your problem isn’t with the driver failing to load, but rather with the driver failing to operate AFTER it has been loaded, then that’s an entirely different issue than the base one. I’d rather take it to a fresh, better named issue with an active OP if at all possible.

Closing this issue wasn’t to say that your problems don’t matter, Id love to help you solve them. It’s just that OP is gone, and further conversation has been, as far as I can tell, off topic. I’m more than happy to continue figuring this out in a relevant thread, though.

kstevens14, 2019-11-17:
> Run this command in your terminal and let me know what it tells you. It will ask for your password because it’s the OS trying to load the kext.
> 
> ```
> sudo kextutil /Library/Extensions/360Controller.kext
> ```
> 
> The reason I say to check the checkbox is because it includes debugging steps to make sure that the driver is correctly loaded when you check the checkbox. If the checkbox can be checked without feedback, then the driver was loaded without a problem. This is a manual check for the same action. If your problem isn’t with the driver failing to load, but rather with the driver failing to operate AFTER it has been loaded, then that’s an entirely different issue than the base one. I’d rather take it to a fresh, better named issue with an active OP if at all possible.
> 
> Closing this issue wasn’t to say that your problems don’t matter, Id love to help you solve them. It’s just that OP is gone, and further conversation has been, as far as I can tell, off topic. I’m more than happy to continue figuring this out in a relevant thread, though.
> 
> Edit: To expand upon this point I’m making, since I’m not sure it was clear. The “ask for permission” is not shown in the preference pane like that. It’s a security block on the signature of the code of the driver. Not on adding the daemon to some accessibility area.

[image]

this is what shows up when I run that command

FranticRain, 2019-11-17:
@kstevens14 That means that the driver was correctly loaded, and that you either approved the driver or (as you did) turned SIP off. So you don't have the same issue as the OP. Or at least the OP's original reason for asking for help is inaccurate. Please make a new issue and fill out the template so I can get a better idea of how to help you.

kstevens14, 2019-11-17:
@FranticRain I created a new issue called wireless xbox controller not identified- issue #974