360Controller/360Controller #941: Controller not working

Piipperi, 2019-10-11:
### Type of Controller
Xbox One

### OS Version
10.15

### Driver Version
1.0.0-alpha.3

### Connection Method
Wired

### Device Name and Info
>Device Name: Power A Spectra Pro Xbox One
<br/>Product ID: 0x542a
<br/>Vendor ID: N/A

### Response of Controller
Controller is recognized by the computer, but not by the driver. Driver just says no devices found

### Response in Preference Pane
Doesn't work

### Detailed Explaination of Issue
Driver not recognizing the controller.

RayJay-UI, 2019-10-12:
Update to Mac OS 10.15 will encounter this failure

ImShalev, 2019-10-12:
I am on macOS 10.15 Catalina, and can confirm controller is recognised by the computer but not by the driver.

Pardonnesmyfrench, 2019-10-21:
Hey there, I have the exact same problem, Idk if the creator will fix this (really hope though) meanwhile, did you find a solution to the issue ?

FranticRain, 2019-10-21:
@Pardonnesmyfrench Use alpha 4. This is fixed.

closed 2019-10-21 by FranticRain