360Controller/360Controller #942: Xbox 360 Controller not supported in Catalina

ImShalev, 2019-10-12:
### Type of Controller
Xbox 360

### OS Version
macOS 10.15 Catalina

### Driver Version
Tested 1.0.0-alpha.3 , 0.16.11 , 0.16.11_Notarized

### Connection Method
Wired

### Device Name and Info
>Device Name: Wired Xbox 360 Controller
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e (Microsoft Corporation)

### Response of Controller
Xbox button just blinks.

### Response in Preference Pane
Device just isn't recognized there. Says "No devices found".

### Detailed Explanation of Issue
Wired Xbox 360 Controller just doesn't work in Catalina, Used to work in Mojave.
The controller just keeps blinking.
I've heard it's because the driver installation does not prompt any kexts authorization.
[image]
[image]

raffaeleg4, 2019-10-12:
Agree totally the same I have

hanhha, 2019-10-12:
+1 I have same issue.

leoma95, 2019-10-12:
Same here. My XboxOne controller is also detected on USB, but 360Controller driver does not detect it in Catalina. Worked fine in Mojave before the upgrade.

LostAccount, 2019-10-12:
> +1 I have same issue.

same here

MaddTheSane, 2019-10-13:
Looking at the Console, I got the following error when I plugged in an Xbox One controller:

````
read - failed to start (0xe00002c2)
````

This was also with version 0.16.6.

FranticRain, 2019-10-14:
That's a `kIOReturnBadArgument`.

macmeister1967, 2019-10-14:
This is the reason you do not install latest new macOS on day of release

raffaeleg4, 2019-10-14:
Indeed. But the update happened automatically without asking unlike the last years

macmeister1967, 2019-10-14:
You mean Catalina? I have Mojave with updates disabled in Software Updates
prefpane, and I am yet to get a notification?!

FranticRain, 2019-10-14:
Looks like Catalina no longer accepts our lazy null options for `inTaskWithOptions` and we have to specify the direction. So for example, this line: https://github.com/360Controller/360Controller/blob/34342c7ba6da60b24f9779f23182f7c58ff79fa8/360Controller/_60Controller.cpp#L550
needs to have that 0 changed to a `kIODirectionIn`.

xshuffl3x, 2019-10-14:
Hello, 
sorry i'm noob in dev so this issue is not fixed ?

MaddTheSane, 2019-10-15:
1. Nobody has fixed the problem code
2. The person who previously signed the kexts has said that he won't be doing it anymore, so someone else would need to sign the kexts.

rom1gal, 2019-10-15:
Same here, LEDs are blinking since the update

briveira, 2019-10-17:
anywhere we can get a compiled release with this patch applied? thanks

SysDragon, 2019-10-17:
FYI it won't work either with the Xbox Elite Controller on last version of Catalina as per the current release

DaBlitzStein, 2019-10-17:
If I knew that didn't upgrade

youngbobby, 2019-10-18:
Ah, this is sad :(

renantiago, 2019-10-18:
Hello guys... any help, please?

FranticRain, 2019-10-19:
Check out [alpha 4](https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.4) and let me know if it works for you all. Looks like rumble is broken, but at least the controller is working now.

ImShalev, 2019-10-19:
@FranticRain did you renew your developer account?

FranticRain, 2019-10-19:
@ImShalev Yes.

ImShalev, 2019-10-19:
@FranticRain Thank you very much!

raffaeleg4, 2019-10-19:
Thank you so much! It's working again

ImShalev, 2019-10-19:
Amazing, closing this issue!

closed 2019-10-19 by ImShalev

renantiago, 2019-10-19:
Thank you guys, you are the best!

d235j, 2019-10-19:
@FranticRain I see that the post-install script still does a `touch /System/Library/Extensions`; this caused install failure for another project I've contributed to. It's supposed to be replaced with `kextcache -i /" but I'm not sure which OS version introduced this feature.

I highly recommend disabling/preventing any potential writes to /System in macOS 10.15 as the new read-only root filesystem will cause those writes to fail.

FranticRain, 2019-10-19:
@d235j Thanks for the heads up, I'll look into it.

leoma95, 2019-10-20:
@FranticRain and team - you guys are amazing. Thank you.

It works fine with Catalina now. I also just installed the first Catalina patch just released by Apple and the controller works fine as well. It just seems that there is a very long delay ˜1min from plugging it in to when it is recognized by the driver configuration app. 

Is there a way to donate $$$ to this team/cause?

FranticRain, 2019-10-20:
@leoma95 I was just recently accepted into the Github sponsors program, so I'll be setting that up soon. Look for it in the coming week. As to why there is some latency between plugin and device recognition, I'm not sure. It could be Apple phoning home to check the notarization of the application before it loads.

PythagoRascal, 2019-11-20:
> Looks like rumble is broken, but at least the controller is working now.

:(, any updates on this or an idea why it's broken?

FranticRain, 2019-11-20:
@PythagoRascal It’s issue that has been a band-aid fix since macOS 10.11 that was made fully impossible in 10.15. At this time, there is no known (to me) solution within macOS to resolve this. Given I have a lot more actual functionality bugs to get to, rumble is very low priority. If you want it in the near future, you’ll need to downgrade macOS versions.

cstches, 2019-12-24:
what about Catalina is now supporting xbox controllers (via bluetooth) ?

mrorato, 2024-01-13:
> Check out [alpha 4](https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.4) and let me know if it works for you all. Looks like rumble is broken, but at least the controller is working now.

Work! Wireless and wired 360 in Catalina 10.15.7