360Controller/360Controller #944: Xbox One Wireless Dongle (discussion)

medusalix, 2019-10-14:
This issue serves as a place to discuss the recently implemented support for the Xbox One wireless dongle. Please report any **problems** and missing **features** in the comments below, so I can add them to this list:

- [ ] Controller rumble
- [ ] Extensions (chatpad, headset)
- [ ] Automatic channel selection (for the dongle)

stevenvachon, 2019-10-15:
If this issue exists with unchecked features, does that then mean that #156 is not completed?

FranticRain, 2019-10-15:
These are all “enhancements.” That was a “new device” issue. They are separate things.

FranticRain, 2019-10-23:
Got another item for the list:
- Update the README to say that wireless xbox one controllers are supported when using a wireless dongle.

stevenvachon, 2019-10-23:
@medusalix are you working on these features as well as part of the bounty, or...? I don't understand.

medusalix, 2019-10-23:
@stevenvachon Yes I do, the rumble support is almost ready (see #950) and I'm currently trying to figure out how the battery status reporting works 😃.

FranticRain, 2019-10-26:
Looks like the "power" button in the preference pane doesn't turn off the controller. Would be a nice to have.

medusalix, 2019-10-26:
@FranticRain I assumed that it would work when the force feedback plugin is loaded. Because the actual `powerOff` function works correctly.

FranticRain, 2019-10-26:
@medusalix Well I'm running signed and notarized builds, so if it was going to work, I'd say that this would do it. Don't know what the problem is.

FranticRain, 2019-11-10:
@medusalix I was trying to test your fix for #967, and I noticed that my older (2013 and 2015 versions, with glossy black plastic) controllers will not pair with the newer dongle. Grabbed a cap with my 2013 controller. 
[OldPair.pcap.txt](https://github.com/360Controller/360Controller/files/3829026/OldPair.pcap.txt)

medusalix, 2019-11-10:
@FranticRain That's good to know. Could you get me a capture on macOS too? I suppose they're pairing correctly but their handshake protocol might be slightly different. Did my fix work when connecting multiple of the newer controllers?

FranticRain, 2019-11-10:
@medusalix Unfortunately, I only have a 2013, a 2015 and a 2018 controller, so I couldn’t test that capability since only one of them matches. I’ll follow up with a macOS packet sniff when I get a chance.

FranticRain, 2019-11-11:
@medusalix Just to be clear, you want a USB packet sniff on macOS of the wireless dongle? Or do you want the console logs? If it's a packet sniff, what is your suggested method of getting one on macOS?

medusalix, 2019-11-11:
@FranticRain Yes I need a USB packet sniff done on macOS. I want to check if it's the association process that's failing or if the handshake needs to be done differently. I'm using Wireshark like [this](https://wiki.wireshark.org/CaptureSetup/USB#macOS) (you might have to try different interface names).

closed 2020-06-08 by medusalix

stevenvachon, 2020-06-08:
@medusalix what of the remaining unchecked features?

medusalix, 2020-06-08:
@stevenvachon I need additional packet captures for those things. Since nobody provided them, I can't implement the missing features.

Juveb, 2020-08-27:
@medusalix 

I was recently looking for XB1 wireless dongle support for my hackintosh when I discovered this thread. Awesome work thus far! I see this is marked as closed, but I'm interested in helping to complete the remaining items if possible. I would especially like to get the wireless headset working.

I posses beginner-medium ability at development, and I am familiar with Wireshark.

stevenvachon, 2020-08-27:
The more that gets done here the better and not just for macOS, but for Linux too which can follow the source code.