360Controller/360Controller #946: Xbox 360 Controller Player not assigned

fwatri, 2019-10-19:
### Type of Controller
Xbox 360

### OS Version
10.15 (Update from 19.10.19)

### Driver Version
360Controller 1.0.0-alpha.4

### Connection Method
Microsoft Wireless Adapter

### Device Name and Info
Device Name: Microsoft Xbox 360 Wireless Receiver for Windows
<br/>Product ID: 0x0719
<br/>Vendor ID: 0x045e

### Response of Controller
The controller actually works just fine

### Response in Preference Pane
Controller is found in the Preference Pane and receives all input

### Detailed Explanation of Issue
After turning on the controller(s) and pressing the connect button on the receiver and controller it "instantly" connects but the controller(s) don't stop blinking (slower pace). I am unable to make the controller(s) assign to a player but otherwise the controller works just fine.

FranticRain, 2019-10-19:
If someone looks into this before I do, best guess is that the `kIODirection...` changes have prevented the LED set packet from actually being sent to the controller. Don't know why that is, but start there.

fiod3s, 2019-10-25:
I can confirm the same issue with the led of a wired Xbox 360 compatible controller. However even my gamepad works fine. I am running macOS Catalina as well.

MoonDogg, 2019-11-03:
I can confirm this as well on exact hardware OP has. MacOS 10.15.1 Game pad works fine, just keeps blinking.

Duxpower, 2021-04-28:
Ey somewhere in the multiverse
Sorry for necroposting. But this might be beneficial to someone.
Found this forum while searching for a fix for the same issue. 

I have 2 controllers. Also, the receiver for pc is a copy, not original. 
Thought it was the receiver at fault first, not letting player 2/controller 2 connect. 

However, if i tried switching batteries, the problem followed just 1 battery. 
(i have 3 batteries). The faulty battery is newly bought.  (gave no indication of low battery life, "kept on blinking").

Anyway, I decided to share this. Maybe someone in the future may benefit from my absolutely mega awesome skills at finding this utterly hard problem to solve.
if you're rich and this solved all your problems and you're now happy instead of sad. 
I know, throw me a bone why don't ya!

Or, should I perhaps find happiness in emojis and thumbs up?

Peace & Understand, or something.
//Duxpower

Keywords:
Xbox 360, controller two not connecting,
one controller not working.
player 2 controler not working.
xbox 360 receiver not working