360Controller/360Controller #947: Can't pair with Wireless Adapter

bestwnh, 2019-10-20:
### OS Version
10.15(19A602)

### Driver Version
360Controller 1.0.0-alpha.4

### Connection Method
Microsoft Wireless Adapter(XBOX ACC, the older version)

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: XBOX ACC
<br/>Product ID: 0x02e6
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
No response at all. Can't pair.

### Response in Preference Pane
The Preference Pane show "No devices found"

### Detailed Explaination of Issue
The the pair led light of wireless adapter is always off. No matter how I press the pair button on the adapter.
BTW, even I connect the Xbox one controller with usb, the Preference Pane still show "No devices found".

FranticRain, 2019-10-20:
@bestwnh Is this an Xbox One or 360 adapter?

It likely needs to be added to the plist, but I don't know if the older dongle follows the same protocol as the newer one.

bestwnh, 2019-10-20:
@FranticRain Thank you for your response. It's an Xbox One adapter. 
I can't find the description on microsoft website now. Here is an [Amazon link](https://www.amazon.com/Microsoft-Xbox-Wireless-Adapter-Windows-Packaging/dp/B07CJ6GVM8/ref=asc_df_B07CJ6GVM8/?tag=hyprod-20&linkCode=df0&hvadid=312127859888&hvpos=1o3&hvnetw=g&hvrand=10887383637098947071&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1013962&hvtargid=pla-568608808999&psc=1&tag=&ref=&adgrpid=61319365399&hvpone=&hvptwo=&hvadid=312127859888&hvpos=1o3&hvnetw=g&hvrand=10887383637098947071&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1013962&hvtargid=pla-568608808999)

How could I help you with that? I'm an iOS developer, if you need any more information or try something just let me know.
BTW, is the 'modern-rewrite' branch ready for build with Xcode 10.3 or Xcode 11 on Catalina? If so, You can tell me which plist I need to add and I may can build the project myself and make a try.

FranticRain, 2019-10-20:
I'm going to hand off to @medusalix on this one. They should be able to help you.

medusalix, 2019-10-20:
@bestwnh The `modern-rewrite` branch can be built with Xcode 10.3. You have to set the `idProduct` property to **`742`** for the driver to detect to your dongle. If it doesn't work, please comment with your [logs](https://github.com/360Controller/360Controller/tree/modern-rewrite#debugging-the-driver) attached.

https://github.com/360Controller/360Controller/blob/4bf049de7b5926036ee3f2f4bc9251cafa96ae7a/WirelessOneController/Info.plist#L42-L43

FranticRain, 2019-10-20:
Actually, I think if you pull from `modern-rewrite` now after I pushed the catalina fixes to it, it should work in Xcode 11.

bestwnh, 2019-10-21:
Hi @medusalix do you have a configuration guide for not code sign? I have an apple developer account, but to code sign the kext it may need to request kext developer id, right? And I follow the configuration in readme but it looks like incorrect. At least the build.sh require to set-up the `DeveloperSettings.xcconfig` to work. But the `DeveloperSettings.xcconfig` values are all for code sign. It made me so confused.
[image]

BTW, I'm use the `modern-rewrite` branch with Xcode 10.3 in Catalina. If the configuration is too complex you can just build a version with `742` idProduct and I can test for you.

medusalix, 2019-10-21:
@bestwnh Try this one: [WirelessOneController.kext.zip](https://github.com/360Controller/360Controller/files/3752980/WirelessOneController.kext.zip)

bestwnh, 2019-10-22:
@medusalix I disabled the signing requirements, and try to run `sudo kextutil /Library/Extensions/WirelessOneController.kext` but keep failing
[image]

medusalix, 2019-10-22:
@bestwnh That's only a warning. If you check with `kextstat` you should see that the kext is loaded.

bestwnh, 2019-10-22:
@medusalix Finally the led light up. But it just keep flashing slowly but never pair the controller.
And the 360controller kext keep unload randomly.(The `Enable Driver` in `Advanced` panel turns off by itself) But WirelessOneController.kext still loaded.

FranticRain, 2019-10-22:
@bestwnh Enable is specific to the wired controller, has no bearing on the wireless driver, is disabled by default unless the wired driver is actively loaded by a connected controller, and is operating as intended.

bestwnh, 2019-10-22:
@FranticRain Got it. But the description is a bit confusing. Better update that before release the official 1.0.0.

FranticRain, 2019-10-22:
@bestwnh Nothing about this driver is ever "official." It's hard to fit good descriptions for everything in that very limited space that is available in the preference pane. I'd actually love to fully remove the preference pane and have a separate, actual application, that takes care of configuration stuff like that.

Maybe one day I'll get the time/motivation to do that.

bestwnh, 2019-10-23:
@FranticRain Sorry I actually mean "formal" 1.0.0.

>  I'd actually love to fully remove the preference pane and have a separate, actual application, that takes care of configuration stuff like that.

Coincidentally I had a similar idea few days ago.

But I'm not very familiar with the Mac app development. And also won't have many devices to test all functions. So I may try to leave the preference pane exist but also rewrite a new Mac app for modern macOS(maybe 10.15+ only) with main function. Then user can use them both.

Is this fits your plan? I'm not very sure do I have enough time to made it, but if it fits your plan I will make a try.

FranticRain, 2019-10-23:
@bestwnh If you’d like to give it a shot, I certainly won’t try to stop you.

medusalix, 2019-10-23:
@bestwnh You may have to try the pairing process multiple times before it works. Also, I need the *logs*.

bestwnh, 2019-10-23:
@medusalix ok I will make more try. But where I can find the logs? The system console logs?

FranticRain, 2019-10-23:
@bestwnh Use Console.app with the search parameters “process:kernel” to filter down to what you’re looking for.

bestwnh, 2019-10-23:
@FranticRain Thx.
@medusalix Here you go the logs. I totally press the button on doge three times. And try to pair the controller about ten times. Still not success.
[kernel.log](https://github.com/360Controller/360Controller/files/3763187/kernel.log)

medusalix, 2019-10-25:
@bestwnh From your logs I can only assume that the old dongle actually  requiresa  slightly different driver implementation than the newer one. Maybe it doesn't support the same WLAN channels the newer dongle uses?

medusalix, 2019-11-06:
@bestwnh I made some minor changes to the driver, please let me know if it works: [WirelessOneController.kext.zip](https://github.com/360Controller/360Controller/files/3816120/WirelessOneController.kext.zip)

bestwnh, 2019-11-07:
@medusalix Thank you. Sorry I don't have time to capture the data in windows recently. I tried the new kext but sadly also not work. I tried multi times. Here is the logs.
[kernel2.log](https://github.com/360Controller/360Controller/files/3819687/kernel2.log)