360Controller/360Controller #948: Alpha 4 hangs Catalina boot! Don't install!

Diamondog36, 2019-10-20:
I tried installing the new alpha 4 version that is supposedly to work on Catalina. It asked me to allow Drew Mills. I did and after the installation finished, it asked me yo restart. Now I can't get back to the login screen, because my Macbook Pro hangs!
Is there a way to uninstall the driver through the command line I can access by pressing Shift on boot? Otherwise my only solution will be a clean reinstall and I don't have time for this!

butaminas, 2019-10-20:
Don't forget that this is an alpha version, not even beta, so bugs are normal.

I had similar issue with an earlier version and fixed it by going to safe mode and deleting all the drivers (look through the issues list, there are several issues explaining how to do this).

This version doesn't make my device hang but it doesn't recognize my xbox one controller as well.

Diamondog36, 2019-10-20:
With the splitting of Macintosh HD in Catalina into HD and HD - Data, none of the solutions in the other issues works. I guess the only solution is the reinstallation, hoping I won't lose any data.

FranticRain, 2019-10-20:
I've been trying to track down this hang bug for the longest time. I don't know what causes it. If you're failing to get in to safe mode/recovery mode, then I'm going to have to suggest that it isn't exactly the driver's fault. In those modes, drivers aren't loaded. Therefore, the things added/changed by 360Controller aren't actually being loaded.

Diamondog36, 2019-10-20:
I managed to solve this by reinstalling Catalina. Nowadays, it happens without formatting the hard drive and without any data loss. However, the installation was unsuccessful. The preferences for this were there, but once I clicked on them, it started in non-dark mode and unable to detect the gamepad. I was using this driver before upgrading to Catalina and when I upgraded, I came across this issue that others also reported. Hence, I was back to square one. I uninstalled the driver for now. But if you figure something out, I will be willing to test it again in the future.

FranticRain, 2019-10-20:
The more recent alphas are definitely dark-mode aware. Did you uninstall the driver through the preference pane before installing the new version, or did you install over your old version?

Diamondog36, 2019-10-20:
I didn't tamper with the driver in any way before upgrading from Mojave (10.14.6) to Catalina. When my Macbook first started after installation, one of the first things I checked was this driver, because there have been compatibility issues in the past. I initially noticed that when I clicked the driver within the preferences, the window goes white (it didn't happen under Mojave) and the driver couldn't detect any gamepads. If I remember correctly, it also defaulted to off for some unknown reason. I re-enabled the driver, but nothing was changing. So, I uninstalled the driver. Upon finishing the uninstallation, I got an error that it didn't uninstall properly, so I restarted the system. Then, I was waiting to see if a new build becomes available, it did, I installed it, I enabled Drew Mills in the Security Preferences as it asked, restarted, and hung. That's how it all happened. I hope it gives you at least a little hint.

FranticRain, 2019-10-20:
So best guess is that you were using a non-1.0.0 version before upgrading, something went wrong with the crappy old method we use to uninstall, and you brought that forward into Catalina with some sort of split-brain state. I'm going to need to make a `uninstall.command` script like I did for `Xone-OSX` so people can actually get this thing uninstalled safely.

The unchecked "enabled" button is to be expected. If the driver isn't loaded, the preference pane has no idea whether it is enabled or not. So it just assumes that it isn't. Had you been able to connect a controller, the checkbox would have been enabled automatically.

Diamondog36, 2019-10-20:
Oh, I completely forgot to mention this and it is important. Good thing your reply reminded me! I was using 1.0.0 alpha 3 before upgrading to Catalina, it wasn't one of the older builds than that.

bestwnh, 2019-10-20:
@FranticRain The alpha-4 support dark mode? I'm also not seeing that feature. And yes I upgrade from the old version(But I forgot which one I used before). And after that I tried uninstalled the alpha-4 and re-install it again, now it still not display correct for dark mode, here is the screenshot with dark mode enabled:
[image]
Any way to make the dark mode theme work fine?

FranticRain, 2019-10-20:
And it wasn’t dark mode aware in Mojave?

bestwnh, 2019-10-20:
Sorry I don't remember the situation on Mojave. I upgrade to Catalina from the first beta version.
And when I open the setting from system preference, it shows up with a dark controller, here is a screenshot during the animation:
[image]
But after about 0.2s it turn to a white one, just like the screenshot I post in the previous comment.

Diamondog36, 2019-10-20:
Sadly, I don't have a screenshot to compare, but in Mojave, it was fine with dark mode. Now in Catalina, it appears exactly like bestwnh's first screenshot. I think we've gone off-topic though. Should another issue be opened for the appearance of the window? Unless it is related to the incompatibility with Catalina.

FranticRain, 2019-10-20:
It's a little off-topic, but I'm grasping at any straw I can find to solve this hang bug, especially since the dark mode preference pane works on my box.

@bestwnh White background and black controller is default state. It swaps background to dark when dark mode is enabled, or the controller to white when it detects light mode. It's three state like that so I can tell what version of the preference pane you are using.

RayJay-UI, 2019-10-21:
In my case, alpha 4 doesn't support dark mode, but it works perfectly.

mclainj, 2019-10-21:
@RayJay-UI Were you on alpha 3 before that? Did you upgrade to alpha 4 before or after updating to Catalina?

RayJay-UI, 2019-10-22:
@mclainj Alpha 3 runs perfectly on Mojave and cannot be run after upgrading to Catalina. After that, I upgrade to Alpha 4 and run it in Catalina.But doesn't support dark mode

visorz, 2019-11-20:
My MacBook Air Mid2012 also hangs on boot after installation of 360Controller 1.0.0-alpha.5. Directly after installation but before reboot I allowed use of "Drew Mills" driver load in security preferences. Even booting into safe-mode pressing shift key does not work.

[image]

I am able to boot into single user mode. How can I remove the driver using terminal commands?

Diamondog36, 2019-11-20:
I searched for a way to remove the driver by Terminal, but Catalina changed the paths the volumes are and in addition to that, the sudo command required to do this no longer works as expected. When I searched, it was too early for any site to have instructions about this kind of problem. Now you may find. But  what I did solve this was a clean reinstall of Catalina. Since Macintosh HD consists of 2 "virtual" partitions (Macintosh HD snd Macintosh HD - Data), the clean reinstall won't delete any of your files and it will let the system boot properly. Once in my desktop, I uninstalled the non-working driver and also looked for files in other folders within the Library yo delete.

visorz, 2019-11-20:
I removed the driver files without reinstalling system:

- hold CMD+R while booting until system recovery dialog appears

- choose terminal app under utilities menu

[image]

- cd /Volumes/MyPersonalHDDNameHere (not the one with ..DATA at the end)
- from here you can start to look for the 360 driver files and delete them. here is the list from uninstall.command script:

1.     \\\"/Library/StartupItems/360ControlDaemon\\\"
2. 	\\\"/Library/Application Support/MICE/360Daemon\\\"
3. 	\\\"/Library/Application Support/MICE/360Daemon.app\\\"
4. 	\\\"/Library/PreferencePanes/Pref360Control.prefPane\\\"
5. 	\\\"/Library/LaunchDaemons/com.mice.360Daemon.plist\\\"
6. 	\\\"/System/Library/Extensions/360Controller.kext\\\"
7. 	\\\"/System/Library/Extensions/Wireless360Controller.kext\\\"
8. 	\\\"/System/Library/Extensions/WirelessGamingReceiver.kext\\\"
9. 	\\\"/Library/Extensions/360Controller.kext\\\"
10. 	\\\"/Library/Extensions/Wireless360Controller.kext\\\"
11. 	\\\"/Library/Extensions/WirelessGamingReceiver.kext\\\"
12. 	\\\"/Library/Extensions/XboxOneBluetooth.kext\\\"

- Please note that I could not find the files:
6. 	\\\"/System/Library/Extensions/360Controller.kext\\\"
7. 	\\\"/System/Library/Extensions/Wireless360Controller.kext\\\"
8. 	\\\"/System/Library/Extensions/WirelessGamingReceiver.kext\\\"
on my system
- Please note that 'XboxOneBluetooth.kext' does not exist, its name is: 'WirelessOneBluetooth.kext'
- Reboot normally and you can log in again

visorz, 2019-11-20:
I went rogue and after a restart to make sure it can reboot again, I downloaded the alpha 5 dmg again and installed it. Before and after the restart no Drew Mills security entry was listet in the security preferences. And as you already guessed my MacBook did not hang on reboot again. I connected my wired Xbox 360 gen 1 controller and it was recognised immediately. 

I think the last install went wrong and blocked my boot procedure because I had a really old version of the driver installed and I guess it was not cleanly removed.

Thank you folks for providing the 360 driver and the support in here!

visorz, 2019-12-18:
1) Yes, I had Oracle's VirtualBox as a program installed. After the Catalina update I did not yet try to run it again.
2) Of course, please share the images.
3) Interesting.
4) Now I understand. Thank you.