360Controller/360Controller #949: hyperkin duke controller not detected

rullinoiz, 2019-10-22:
### Type of Controller
Xbox One

### OS Version
10.15 (Catalina)

### Driver Version
version 1.0.0 alpha 3
version 1.0.0 alpha 4

### Connection Method
wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: hyperkin duke
<br/>Product ID: 0x1618
<br/>Vendor ID: 0x2e24

### Response of Controller
it just vibrates when i plug it in, no other response

### Response in Preference Pane
the pane doesnt see it

### Detailed Explaination of Issue
the controller vibrates when i plug it into my computer, so it is getting power, however the drivers dont see it as an xbox controller. its branded by microsoft and is listed as an xbox one controller that works with windows 10 and xbox one. macOS sees this controller, and i have tried different cables that have worked with other wired controllers that the drivers saw.

FranticRain, 2019-11-12:
Here's a build with this controller added. Let me know if you have any issues with it. All future builds will include this controller.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3834382/360ControllerInstall.dmg.zip)

closed 2019-11-12 by FranticRain

rullinoiz, 2019-11-14:
> Here's a build with this controller added. Let me know if you have any issues with it. All future builds will include this controller.
> 
> [360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3834382/360ControllerInstall.dmg.zip)

wow, thanks. i didn’t think you would add it just for me, but really appreciate it

FranticRain, 2019-11-14:
Adding new controllers is just part of maintaining the driver. Any and all Xbox-compatible devices are welcome. 👍