360Controller/360Controller #951: Dark mode incompatible with Catalina

bestwnh, 2019-10-22:
This issue is start from https://github.com/360Controller/360Controller/issues/948 But after that I make some test and get a summary:
The issue is not related upgrade the driver form old version or update the system from old version.
I use a new install Catalina with version alpha-4 can still reproduce the issue. The key point is System Integrity Protection(aka `SIP`) status.
- When SIP is disable, the dark mode theme work fine.
- When SIP is enable, the dark mode theme not show.

@FranticRain Hope these information can help you to solve it.

FranticRain, 2019-10-22:
@bestwnh That explains why it works on my box. Best guess is that the cert I use to sign the prefpane is wrong or invalid, whereas the kext one works fine.

FranticRain, 2019-10-31:
Looks like (when SIP is enabled) the preference pane goes through multiple draw calls. One with dark mode, then one with light mode. So we end up with the night mode. Looks like this is a Catalina change any may be something we can't fix. May have to swap to an application and stop using the preference pane.

FranticRain, 2019-11-10:
[Confirmation of issue.](https://www.noodlesoft.com/blog/2019/08/28/preference-panes-and-catalina/) Looks like it's an Apple bug. Definitely going to have to make a separate configuration app priority. Most difficult part will be passing settings to/from the driver to a user-space application.

bestwnh, 2019-11-11:
@FranticRain The "settings" you mean is the button mapping configuration? Why it would be the most difficult part?

FranticRain, 2019-11-11:
@bestwnh Settings includes all configurations from swapping sticks to pretending to be a 360 controller. This is the hardest part because there is no known (to me) official way to pass settings to a driver. We were cheating a bit by having our UI in the same kernel space via a preference pane. We’d lose that entirely in a user space application. That isn’t to say that the current settings solution is perfect. It’s just to say that it exists.

bestwnh, 2019-11-17:
@FranticRain I found a way to save and load the configurations. I will try to make a demo app and give you to have a try within this week.

bestwnh, 2019-11-21:
@FranticRain Here is a demo app.(Maybe need to allow in Security & Privacy) Sometime may crash. But just reopen it.It will show the first controller input and contain some configurations button for you to try.
[DemoApp.app.zip](https://github.com/360Controller/360Controller/files/3875789/DemoApp.app.zip)
[image]

FranticRain, 2019-11-22:
@bestwnh Looks pretty good. Looks like the settings are a bit inconsistent, though. For example, none of the right stick inversion options worked until I swapped the sticks. And some of the settings didn't activate/deactivate whenever I checked/unchecked the checkboxes.

I assume you're just editing settings that this driver uses and this isn't a DriverKit app. I appreciate the effort, but there are some bugs with the current settings system that I'd like to resolve as part of this move to the user space. For example, some of the settings don't apply when the driver first starts up. For example, open your app and enable any of the settings. Disconnect the controller and re-connect it. You'll notice that the setting is not applied to the controller. This is because it's impossible for the driver to load these settings the way it currently does.

There are few solutions to this problem: 1. Stop using a kext. 2. Switch to a different settings scheme. 3. Update the daemon to somehow re-prompt the driver to load the settings shortly after connection.

Option 3 is probably the best option that resolves your issue, but I'd need to verify that it's possible before we continue on with this settings scheme.

bestwnh, 2019-11-23:
@FranticRain Some of issue you point out should be my bad. Because I'm not complete the code yet. It's just a demo to confirm the save/load configuration function works well. I totally rewrite the code in Swift so it may have some bugs. But I use the exactly same function `IORegistryEntrySetCFProperties` and `CFPreferencesSetValue` to do the configuration work. So it could do the exactly same thing as the exist prefpane version.

I'm not familiar with kext nor DriverKit, so the only thing I can do now is follow the old logics then improve them in my way. Is it possible to make the driver with DriverKit not kext? If so a rewrite would be way better for long term.

FranticRain, 2019-11-23:
@bestwnh DriverKit wraps drivers up in user-space applications instead of using kexts. There's some concern that it won't be compatible with older games that use the driver poorly, but DriverKit is the future. https://developer.apple.com/documentation/driverkit

bestwnh, 2019-11-23:
@FranticRain Agree! And most old game already kill by "64bit only" on Catalina. So DriverKit should be good enough. But because I have almost no experience with driver and USB/HID device. And there is almost no resources on the internet to introduction things like this(The DriverKit document is just list the class without detail information). Can you point out the start point for me? The best way is a very simple function to help me sort out the basic communication process. Just like the `startDevice` function in `Pref360ControlPref.m`. Then I can implement a demo app and actually run the process. The current 360Driver project is too complex for me. I don't really know where to start.

BTW, is that mean I also can re-implement the demo app with DriverKit? Or I need to make a DrierKit base driver for the controller first?

FranticRain, 2019-11-23:
@bestwnh https://developer.apple.com/documentation/usbdriverkit should be a better starting point. You would have to completely start from scratch. The settings work you’ve done would not translate to this new model, since everything is handled in user-space.

bestwnh, 2019-11-24:
@FranticRain The document of Apple DriverKit is all "No overview available.". So maybe the right direction for me is continue finish the configuration app for the first step. Goodbye driver land 😄

FranticRain, 2020-01-02:
Swapping this one over to "wontfix" since it's a confirmed Apple bug and outside of our ability to fix.

bestwnh, 2020-01-02:
@FranticRain Just for notice my app still WIP and hoping release within 2 months. It can "fix" this  issue😄. And BTW do you have any way to solve the ForceFeedback issue?

FranticRain, 2020-01-02:
@bestwnh It looks like it’s never coming back. Full functionality won’t be restored in 10.15 and later until we make a DriverKit app. I’ve gotten the ball rolling by requesting a system extension entitlement from Apple, but have not been granted one yet.

bestwnh, 2020-01-03:
@FranticRain So you really going to make a DriverKit app? That's fantastic. Can't wait for it.