360Controller/360Controller #952: Xbox Controller Model 1708 not working in macOS Catalina

jtsolano, 2019-10-22:
### Type of Controller
Xbox One Controller ( Model 1708 )

### OS Version
macOS Catalina ( 10.15 )

### Driver Version
360Controller 1.0.0-alpha.4
360Controller 1.0.0-alpha.3
360Controller 0.16.10
360Controller 0.16.4

### Connection Method
Wired ( USB Type A to Micro USB )

### Device Name and Info
>Device Name: Wireless Controller for Xbox One Model 1708
<br/>Product ID: 0x02FD
<br/>Vendor ID: 0x045E

### Response of Controller
I first connect one end of the cable ( USB Type A ) to the Mac , then I proceed to connect the other end of the cable ( Micro USB ) to the Xbox controller. When I connect it the Xbox controller, it vibrates once and lights remain off throughout the whole process. Also at the beginning of this process the Xbox controller is off.

### Response in Preference Pane
The panel shows always "No devices found". I have tried to connect the controller again after disabling the driver and re-enabling it. I have also tried rebooting the Mac with the driver disabled and enabled with no success.

### Detailed Explanation of Issue
My Mac's bluetooth is turned off and I am trying to use a Xbox One Wireless Bluetooth Controller with a wired connection on my Mac. After connecting this Xbox controller to the Mac, it vibrates once and lights never turn on at any moment.

After connecting the cable to the Mac and the Xbox controller, if I press the Xbox button for a while ( the button that lights up when pressed ), the button starts emitting a pulsating light until I disconnect the cable.

Also, after connecting the cable to the Mac and the Xbox controller, I have checked if the Mac detected any new USB connections but no new connections seemed to appear. I checked this at [About This Mac -> System Report -> Hardware -> USB -> USB Device Tree].

FranticRain, 2019-10-22:
Have you ever connected the controller to your computer via Bluetooth?

Also, please run this command in your terminal. It will ask for your password, but all it is trying to do is manually load the driver.

```bash
sudo kextutil /Library/Extenstion/360Controller.kext
```

jtsolano, 2019-10-22:
Hello FranticRain,
Yes, I have connected previously the mentioned Xbox controller using bluetooth to the Mac. Can connecting previously the controller via bluetooth affect the wired connection?

Here are also the results of the command:
sudo kextutil /Library/Extensions/360Controller.kext
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).

FranticRain, 2019-10-22:
It's been suggested that connecting a controller via Bluetooth breaks the ability to use the controller as a wired device. I was pretty sure that it was just the Bluetooth connection taking priority over the wired one, and have confirmed this through my testing. However, using this exact same controller, I was unable to reproduce your issue.

Alpha 4 should work, and it's the only one that should work. All other versions are guaranteed not to work.

jtsolano, 2019-10-22:
I see, thanks for the help FranticRain.
Just to be sure I understand what you're saying, you tried the exact same model as the one I mentioned and you used a wired connection, and you didn't have any issues with it. All buttons, analog sticks and triggers worked fine. Is this correct?

FranticRain, 2019-10-22:
Correct. I even connected it via Bluetooth, turned the Bluetooth off, and then plugged it in.

d235j, 2019-10-22:
Could this possibly be an issue related to controller firmware? When connected to an Xbox or to Windows it is possible that firmware gets updated, and this has solved problems for others in the past.

FranticRain, 2019-10-22:
That's a fair point. It absolutely could be a firmware issue. I would find it highly surprising that firmware was an issue with a 2018 controller, though.

@jtsolano Do you have an Xbox One console or Windows PC that you can use the update the firmware of the controller?

jtsolano, 2019-10-22:
I do have a Windows PC although I did connect it recently to check if the triggers of this model were working with a Windows PC.
This is actually the reason why I am trying to use your drivers with a wired connection on my Xbox Bluetooth controller, the triggers don't work over bluetooth but I am able to connect it through bluetooth and see the firmware version in the System Report panel. I can confirm that the firmware version of my Xbox Bluetooth controller is 0x0903.
I will try to update the firmware through my Windows PC but I would appreciate if you could confirm me which is your firmware version of your Xbox Bluetooth controller.
Also, out of curiosity, do your triggers work over bluetooth on your Mac with the Xbox Bluetooth Controller Model 1708?

FranticRain, 2019-10-22:
I used the latest version of the firmware on my controller (as reported by the "Xbox Companion" app) 3.1.1221.0. Do not trust that number that Apple reports as the firmware number.

I can also confirm that my triggers work as expected. Seems like your controller may be busted.

jtsolano, 2019-10-23:
Hello FranticRain,
I used the "Xbox Accessories" app to check the firmware of my Xbox One Bluetooth Controller, its also 3.1.1221.0 and no updates where available. I also know for sure that my triggers are not busted because they work perfectly fine on Windows PC. I have been using this webpage to test that buttons analog sticks and triggers work properly in Windows and Mac: https://html5gamepad.com

I'm sorry to be so persistent but I just want to make sure. You connected a Xbox One Wireless Bluetooth controller over bluetooth ( no wired connection ) to you Mac and your triggers were responding? When I try this, the webpage that I mentioned above and the games I play do not respond to triggers. Could I please ask you to try the webpage I mentioned or know how do you check that the triggers work on your side?

I am starting to think I may be missing something on the Mac side, maybe the bluetooth firmware isn't updated.
Again, thanks for all the help, really appreciate it.

FranticRain, 2019-10-23:
Not only did I use the exact same controller with the exact same firmware on the exact same version of macOS via Bluetooth, I also used that exact website to check. It shows up under the buttons as an analog value between 0 and 1, but it was definitely working as expected.

jtsolano, 2019-10-23:
There is a possibility I may not have uninstalled one of the 360Controller drivers before installing another one. Could it be possible that installing multiple versions without uninstalling a previous one could have messed up something?

Is there a way of cleaning and uninstalling everything to perform fresh install of the 360Controller drivers? If this doesn't work I'll start considering reinstalling the whole OS.

I am also not sure how to check if this step mentioned to install the 360Controller 1.0.0-alpha.4 has been performed correctly. The step states the following:

`Make sure to allow access to the developer in System Preferences > Security & Privacy`

Is there a way to check the current developers that have been allowed access? In my _Security & Privacy_ settings I have the _App Store and identified developers_ option selected but it doesn't show anything else. I remember clicking on a _Allow_ button and allowing a developer named "Dred Mills" shortly after installing the drivers. Is this what those instructions refer to?

jtsolano, 2019-10-23:
I just tested the 360Controller 1.0.0-alpha.4 in another Mac which is the same model as mine ( Mac Mini ) with another Xbox One controller also model 1708. After updating the other Mac to Catalina Os, I proceeded to install the 360Controller 1.0.0-alpha.4 drivers and then I connected over bluetooth the Xbox One controller. The response in the preference panel of 360Controller kept showing "No devices found" but in the Mac's bluetooth panel the controller appeared as connected ( same as in the first Mac ). The firmware of the other Xbox controller is also updated. 

At this point I am not sure what could be causing the issue after testing in two separate Macs and controllers. Could you please send me two code number of your Xbox Controller? The one above and bellow the barcode behind the battery compartment? I just want to compare these numbers and see if the controllers were made in the same batch, maybe mine is older.

Thanks again for all the help FranticRain.

FranticRain, 2019-10-23:
The preference pane will only see controllers that are using the driver. Xbox One controllers are natively supported in macOS, and do not use the driver. Therefore, they will not show up in the preference pane. This is working as expected. You are not using the driver unless you are connected via a wire or via the wireless dongle. This is mentioned in the README: https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-bluetooth

jtsolano, 2019-10-23:
Hi FranticRain,
I am sorry for the misunderstanding. I wasn't aware that the drivers didn't support bluetooth connection itself. 
I have also discovered that my Xbox Controller does actually work fine. Apparently the issue was that I was using Safari to browse the site we used to test our controllers. It turns out that using Chrome gives different results. The all the controller's buttons, triggers and analog sticks worked fine on that website when using Chrome.
Again, sorry for the misunderstanding and thanks again for the help.

closed 2019-10-23 by jtsolano

FranticRain, 2019-10-23:
No problem. Glad to know that you figured out, and the issue with Safari are duly noted. I’ll see if I can verify the issue and reach out to Apple to see if they know anything about HTML gamepad issues in Safari.