360Controller/360Controller #953: Cannot Load Driver - unknown error, code: -603946986

xorxornop, 2019-10-23:
### Type of Controller
Xbox One (do not have others to compare, if applicable)

### OS Version
10.14.2 Mojave

### Driver Version
1.0.0-alpha.4

### Connection Method
Wired

### Device Name and Info
>Device Name: Controller
<br/>Product ID: 0x02dd
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Controller vibrates upon connecting it, but the X button ("guide button") does not light up. Any attempted inputs have no effect.

### Response in Preference Pane
When attempting to enable the driver via the 'Enable Driver' checkbox, the following error is displayed via a popup messagebox: "The driver has failed to load due to an unknown error, code: -603946986. Please report this error to the driver maintainer." with title "Cannot Load Driver".
I uninstalled the driver using the 'Uninstall' button, and reinstalled it, but the problem persists unfortunately.

### Detailed Explaination of Issue
There isn't much more to this issue than the above description, other than to mention that I was previously using driver version 0.16.11 successfully (same OS version, same connection method, etc.), so I am sure that the controller is not at fault. The only thing that changed is moving from 0.16.11 to 1.0.0-alpha.4.

FranticRain, 2019-10-23:
I'll have to do some more digging to make sure that it's the right error code, but that code should correspond to `kOSKextReturnInUse`. Which would mean that the driver is already loaded into memory, and that your controller just isn't responding to it. Can you run this in your terminal and let me know what the output is, to make sure that the driver is working as expected? It will prompt you for your password, but this is just a macOS utility to load drivers manually.
```
sudo kextutil /Library/Extensions/360Controller.kext
```

plasticlobster, 2019-10-23:
I have the same issue. Here's the output from my kextutil run.

`
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/System/Library/Extensions/1C207F9B-91AD-403A-8D88-D8B3EDA3B7C2.kext
Bundle (/System/Library/Extensions/SteamInputX.kext) failed to validate, deleting: /Library/StagedExtensions/System/Library/Extensions/1C207F9B-91AD-403A-8D88-D8B3EDA3B7C2.kext
Unable to stage kext (/System/Library/Extensions/SteamInputX.kext) to secure location.
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV9IOService' for vtable '__ZTV17Xbox360Peripheral' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22Xbox360Pretend360Class was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOneControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOnePretend360Class was not patched because its parent, the __ZTV22XboxOneControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV27XboxOriginalControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) Can't load kext com.mice.driver.Xbox360Controller - link failed.
(kernel) Failed to load executable for kext com.mice.driver.Xbox360Controller.
(kernel) Kext com.mice.driver.Xbox360Controller failed to load (0xdc008016).
(kernel) Failed to load kext com.mice.driver.Xbox360Controller (error 0xdc008016).
Failed to load /Library/StagedExtensions/Library/Extensions/360Controller.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).
`

plasticlobster, 2019-10-23:
I'm not the *right* kind of developer here, so forgive me if I'm absolutely wrong...
https://github.com/360Controller/360Controller/blob/master/360Controller/Info.plist#L447

But shouldn't the `CFBundleIdentifier` be set to `com.mice.driver.Xbox360Controller`?

FranticRain, 2019-10-24:
No. That line is correct. That file tells the driver what classes to load when it finds a device. The chatpad isn't a controller, so it doesn't use the Xbox360Controller driver, it uses a different IOHIDEventDriver instance to handle things.

Second, to be clear, because we get a lot of people who dogpile on issues saying "I have the same issue" for literally anything that is broken: Do you also get an identical error code when attempting to check the "enable driver" checkbox?

Third, that load is basically telling you that 360Controller failed to load because whatever the heck `1C207F9B-91AD-403A-8D88-D8B3EDA3B7C2.kext` and `SteamInputX.kext` are, they aren't signed correctly.

Fourth, are you using a Hackintosh, because this has all the telltale signs of one.

Fifth, do you experience this issue with alpha 3, since (if you have the same issue) you are running Mojave.

HitoriSensei, 2019-10-26:
For me `alpha.4` also throws
```
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV9IOService' for vtable '__ZTV17Xbox360Peripheral' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22Xbox360Pretend360Class was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOneControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOnePretend360Class was not patched because its parent, the __ZTV22XboxOneControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV27XboxOriginalControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) Can't load kext com.mice.driver.Xbox360Controller - link failed.
(kernel) Failed to load executable for kext com.mice.driver.Xbox360Controller.
(kernel) Kext com.mice.driver.Xbox360Controller failed to load (0xdc008016).
(kernel) Failed to load kext com.mice.driver.Xbox360Controller (error 0xdc008016).
Failed to load /Library/StagedExtensions/Library/Extensions/360Controller.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).
```
when I try to load it.

`alpha.3` works fine

**Type of Controller**
Xbox One

**OS Version**
10.13.6

**Connection Method**
Wired

FranticRain, 2019-10-26:
Yeah, I know what the issue is and will be releasing a fixed version soon. Just need to wrap up a laundry list of things before I do.

AXAXAXAXAXAXAXAX, 2019-10-29:
**Cannot Load Driver**
> The driver has failed to load due to an unknown error, code: -603946986. Please report this error to the driver 

And I ran "sudo kextutil /Library/Extensions/360Controller.kext" in terminal:
```
Last login: Tue Oct 29 11:07:00 on console
Juns-Mac-Pro:~ x$ sudo kextutil /Library/Extensions/360Controller.kext
Password:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Can't open CFBundle for /System/Library/Extensions/roller.kext.
Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 0.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

Can't create kext from cache: /System/Library/Extensions/roller.kext is already open and has a different CFBundleIdentifier from identifier->path cache entry 0.
Disabling KextAudit: SIP is off
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV9IOService' for vtable '__ZTV17Xbox360Peripheral' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22Xbox360Pretend360Class was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOneControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOnePretend360Class was not patched because its parent, the __ZTV22XboxOneControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV27XboxOriginalControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) Can't load kext com.mice.driver.Xbox360Controller - link failed.
(kernel) Failed to load executable for kext com.mice.driver.Xbox360Controller.
(kernel) Kext com.mice.driver.Xbox360Controller failed to load (0xdc008016).
(kernel) Failed to load kext com.mice.driver.Xbox360Controller (error 0xdc008016).
Failed to load /Library/Extensions/360Controller.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).

```

**Type of Controller**
Xbox One (Wireless adapter version)

**OS Version**
10.14.5 (Hackintosh)

**Connection Method**
Wired

FranticRain, 2019-11-04:
This should be resolved in alpha 5. Let me know if it works.

AXAXAXAXAXAXAXAX, 2019-11-05:
Driver can be enable in preference pane now! No error showed. But can't find devices after wired connecting my Xbox One controller. However "Controller" showed in System info > USB > USB3.1
Mine is a hackintosh system. USB kexts(USBInjectAll.kext, GenericUSBXHCI.kext) are injected. Other usb2.0/3.0 devices works fine so far(including ps4 controller).
Ps: My Xbox One controller works very well in 10.15(iMac late 2017, non hackintosh) with Alpha 4 and 5.

**Type of Controller**
Xbox One (Wireless adapter version)

**OS Version**
10.14.5 (Hackintosh)

**Driver version**
Alpha 5 

**Connection Method**
Wired

etyrnal, 2019-11-05:
Same error code when trying to Enable Driver (**No Devices Found**)

**Type of Controller**
Xbox 360 (rock candy --- Bus 058 Device 001: ID 0e6f:011f 0e6f Rock Candy Gamepad for Xbox 360  Serial: [redacted] )

**OS Version**
10.11.6 El Capitan

**Driver Version**
1.0.0-alpha.4 **and** Version 1.0.0-alpha.5

**Connection Method**
Wired

**Device Name and Info**
**Device Name**: Controller

**Product ID**: 011f ("Rock Candy Gamepad for Xbox 360")

**Vendor ID**: 0e6f (rock candy)

**Response of Controller**
X button ("guide button") does light up 100% for abotu 1 second, then goes completely dark. Any attempted inputs have no effect.

**Response in Preference Pane**
When attempting to enable the driver via the 'Enable Driver' checkbox, the following error is displayed via a popup messagebox: "The driver has failed to load due to an unknown error, code: -603946986. Please report this error to the driver maintainer." with title "Cannot Load Driver".
I uninstalled the driver using the 'Uninstall' button, and reinstalled it, but the problem persists unfortunately.  I did this for alpha 4 and 5

puter:~ moi$ sudo kextutil /Library/Extensions/360Controller.kext
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV20ChatPadKeyboardClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The super class vtable '__ZTV11IOHIDDevice' for vtable '__ZTV22Xbox360ControllerClass' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22Xbox360Pretend360Class was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOneControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV22XboxOnePretend360Class was not patched because its parent, the __ZTV22XboxOneControllerClass, was not patchable.
(kernel) kxld[com.mice.driver.Xbox360Controller]: The __ZTV27XboxOriginalControllerClass was not patched because its parent, the __ZTV22Xbox360ControllerClass, was not patchable.
(kernel) Can't load kext com.mice.driver.Xbox360Controller - link failed.
(kernel) Failed to load executable for kext com.mice.driver.Xbox360Controller.
(kernel) Kext com.mice.driver.Xbox360Controller failed to load (0xdc008016).
(kernel) Failed to load kext com.mice.driver.Xbox360Controller (error 0xdc008016).
Failed to load /Library/Extensions/360Controller.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).
puter:~ moi$ 

anything i can do?  or does this require fix on your end?  I'm going to return the controller for now since i don't know how long or if this might take to fix.  I'll wait a couple days tho, if you need me to try anything.

FranticRain, 2019-11-05:
@AXAXAXAXAXAXAXAX So you're saying that wired doesn't work on your machine, but the driver supposedly works on your machine? What modes are you using it in that work correctly? Bluetooth? Wireless adapter?

@etyrnal You're exactly the use case I was looking for. Based on the failures of alpha 4, I assumed that alpha 5 would only work on 10.14+. I can get a build out to you in the next couple of days that should work with 10.11. But I can't promise that it'll work with your controller. Rock Candy controllers are known to be problem children when it comes to working with this driver.

AXAXAXAXAXAXAXAX, 2019-11-05:
@FranticRain I connected my controller using wire. Those problems occurred before I deleted kexts. The controller works very well now.

FranticRain, 2019-11-05:
@AXAXAXAXAXAXAXAX Oh okay. I'm going to go ahead and say that fighting with the intricacies of a hackintosh install is the responsibility of the user and outside the scope of support of the driver. Glad you got it working.

etyrnal, 2019-11-05:
> @AXAXAXAXAXAXAXAX So you're saying that wired doesn't work on your machine, but the driver supposedly works on your machine? What modes are you using it in that work correctly? Bluetooth? Wireless [...]

@FranticRain do you need anything from me?  what can i do to help?  or do i just wait?

FranticRain, 2019-11-05:
@etyrnal Nothing for you yet. When I get a chance to build the driver for you, I'll need confirmation that it works so that I can confirm my suspicions.

FranticRain, 2019-11-06:
@etyrnal This is a super stripped down version just to see if this works, but it should work in 10.11. Let me know how it works.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/3812654/360ControllerInstall.dmg.zip)

etyrnal, 2019-11-06:
@FranticRain  -- i will hopefully be able to try this when i get home this evening.  Is this expected to work for WIRED and wireless?    My controller specifically is wired.  Just want to make sure.  What info do you want me to collect besides a pass/fail experience report?

FranticRain, 2019-11-06:
@etyrnal My main concern is getting you working. So this should work for both wired and wireless, but my only interest is whether or not it works for your scenario. If it works, I’d like confirmation that it works. If it doesn’t, then I’d like the same debug information you’ve given, but for this new driver. That’s it.

etyrnal, 2019-11-07:
@FranticRain  Good news.  Uninstalled alpha 5.  Installed this test version you sent.  Rebooted.  Plugged in my rock candy xbox 360 controller.  For about 250ms ALL indicator lights on controller light up.  Then top left light lights up steady and stays lit.  System preference for "Xbox 360 Controllers", first tab, controller test.  Every button press light the appropriate button in the panel.  All of the invert controls function as expected.  Dead-zone adjustments work as expected with one exception.  Setting the dead-zones to maximum value (slider all the way to right) makes them behave as if there were no dead-zones (same as all the way left).  Have no rumble -- cannot test it.  Under Binding, all functions work logically.  Advanced tab, all is logical/working.

Only weirdness i found, which feels like is not related to your driver, is that in MAME (run under wine installation) DOES see inputs from controller for mame menus -- even setting player inputs for a machine.  But once game launches, no inputs seem to get to the game itself.

i hope this helps.

etyrnal, 2019-11-07:
@FranticRain (and for anyone else who's interested)

I was able to get my newly purchased rock candy ($15 walmart) Xbox 360 controller to work in MAME .215, under macOS 10.11.6 (El Capitan) by downloading THIS MAME .215 ( http://sdlmame.lngn.net/mame0215-64bit.zip ) from this page ( http://sdlmame.lngn.net/ ), which needed SDL2 (which would NOT install using brew install SDL2, due to my OLD Xcode, or an error in the brew install script), but i did find SDL2 for macOS at this site ( https://www.libsdl.org/download-2.0.php ) by downloading this .dmg ( https://www.libsdl.org/release/SDL2-2.0.10.dmg ), and following the directions to copy the "SDL2.framework" folder contained in the mounted .dmg, into the "/Library/Frameworks" folder.

in my MAME .215 folder i created a link to the roms folder (elsewhere, on shared nfs network volume)...  

i run MAME by opening terminal and:

[redacted]:mame0215-64bit_macOS [redacted]$ /Volumes/[redacted]/[redacted]/mame_all/MAME_215/mame0215-64bit_macOS/mame64 

AWESOME!  And the controller works GREAT!

FranticRain, 2019-11-07:
Perfect. Thanks for your help. I'll see what I can do for the next release to make this a more permanent solution.

closed 2019-11-07 by FranticRain

sfxman, 2019-11-28:
Hi, I just wanted to let you know that I had the same issue (Running 10.11.6, error code -603946986 trying to start the driver, very similar output from kextutil) and the latest .zip file you provided here has cleared things up for me too, the controller works perfectly now. Thanks for this great driver!

aj9777, 2020-01-01:
I downloaded alpha 5 version, but still not solved, the controller vibrate but no device found

MAC OS 10.11.6

same error code

FranticRain, 2020-01-01:
@aj9777 Have to tried the version in this thread, https://github.com/360Controller/360Controller/files/3812654/360ControllerInstall.dmg.zip

I think this will probably solve your problem.