360Controller/360Controller #954: Elite Black controller not detected

Dragenfi, 2019-10-23:
### Type of Controller
Replace this text with: Xbox One Elite Black

Mojave

### Driver Version
Alpha4

### Connection Method
Wired via usb

### Device Name and Info
 N/A 

### Response of Controller
If I turn it on before connecting via usb it vibrates once. If I don’t turn it one before hand, nothing happens 

### Response in Preference Pane
Comes up with no devices detected 

### Detailed Explaination of Issue
My Elite black controller is not being detected when I connect it via usb. Is there an issue with the elite black controller?

FranticRain, 2019-10-23:
You must update your original post with your controller information. A guide on how to do so can be found here: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id.

Controller information is only not required for issue that do not involve controllers. For example, a feature request that isn't specific to any controller, or maybe a typo somewhere in the code or documentation. For all issues individuals have with getting a controller working, controller information is ABSOLUTELY VITAL to solving your problem. Not only does it let us know if your controller is already supported, it also lets us know that the operating system can see the controller and is a crucial step in resolving your issue.

Please notify my when you have updated your original post with your controller information, or let me know if you are unable to find it.

MrO007, 2019-11-01:
Hello,

I'm having a similar issue with Elite White.

Product ID:	0x02e3
  Vendor ID:	0x045e  (Microsoft Corporation)

Thanks

FranticRain, 2019-11-02:
@MrO007 Can you check the “enable driver” checkbox, and follow any debug information it gives you and let me know if you have any further issues?

FranticRain, 2019-11-04:
Can you confirm that this issue persists in alpha 5?

MrO007, 2019-11-04:
Hello, issue resolved by clicking “Enable Driver”.  Initially, the app complained that the driver wasn’t installed.  So I reinstalled the package.  On restart, Enable Driver was already ticked and the controller worked flawlessly.  

Should the readme.md be updated with making sure “Enable Driver” is checked?  At the time, when it was unchecked, I just thought it was a special thing that had another purpose.

FranticRain, 2019-11-04:
Your original understanding of the "enable driver" button was correct. I simply added the debug functionality to the "enable driver" checkbox because it was simple and easy to do as part of the load process. Also because a lot of people who attempt to debug the issue for themselves check that checkbox as part of their debugging process. That step has been added to the README and issue templates of the `modern-rewrite` branch that this code is built on. I would rather not update the stable line instructions with instructions specific to the alpha version.

Since the actual issue here is resolved, I'm going to close this down.

closed 2019-11-04 by FranticRain

Dragenfi, 2019-11-04:
Sorry for delay in updating my controller information, I have been unwell. I will try to reinstall the app and see if that resolves my issue.

MrO007, 2019-11-04:
I think we both had the same controller with the same symptoms (doesn't mean same problem, of course).  However, I found the "Enable Driver" unchecked and when I checked it I got "Driver Missing".  Reinstall resolved it.

Dragenfi, 2019-11-04:
I have installed 360Controller 1.0.0-alpha.5 and ensured that Enable Driver was checked and its now working perfectly!

Thanks MrO007 and FranticRain