360Controller/360Controller #955: Can't pair with Wireless Adapter

wagenet, 2019-10-24:
### Type of Controller
Wireless Xbox 360

### OS Version
10.15 (19A602)

### Driver Version
1.0.0-alpha.4

### Connection Method
Wireless

### Device Name and Info
  Product ID:	0x0291
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.07

### Response of Controller
When trying to pair it does the spinning lights then goes back to the slow flash of all lights.

### Response in Preference Pane
No response

wagenet, 2019-10-24:
I should note that this used to work on 10.14.

FranticRain, 2019-10-24:
What version of the driver were you using in 10.14?

wagenet, 2019-10-24:
1.0.0-alpha.3

WaddlesM, 2019-10-26:
Well im thinking that the update can't find the 3 party adapter making it not pair. Or that it isn't compatable with your system?

FranticRain, 2019-10-26:
@WaddlesM I wouldn't make any assumptions like that. Without further testing from a developer, it's hard to say anything specific about what has changed between the versions of the driver and the operating system. I think it's something that needs to get fixed in the driver that was changed in Catalina, but I'm not going to make any wild assumptions.

wagenet, 2019-10-28:
@FranticRain I'm a software developer (though not macOS) so I'm comfortable with poking around a bit if there's anything useful you think I could unearth.

FranticRain, 2019-10-28:
@wagenet I just need to carve out the time to investigate. Shouldn't be too hard. Since I stopped working on 360Controller, I've picked up quite a few other things that are now eating all of my time. Finding time for this driver is tough, but I'm trying to cram it in there. That isn't to say that you can't take a look, but navigating the macOS kernel is a pretty daunting task no matter your level of software development, and I can't recommend it.

wagenet, 2019-11-06:
@FranticRain by the way, I just tried alpha5 which unfortunately does not appear to resolve this issue.

FranticRain, 2019-11-06:
@wagenet I didn’t expect it to. I just haven’t gotten a chance to chase this one down yet.

tchan, 2019-12-21:
Unfortunately having the same issue.

mihalski, 2020-01-26:
I can confirm I'm having the same issue.
Works fine under linux if that's of any value. The pairing is successful at the `input` line.
```
[74425.976469] usb 1-1.1: new full-speed USB device number 11 using dwc2
[74426.081504] usb 1-1.1: New USB device found, idVendor=045e, idProduct=0291, bcdDevice= 1.07
[74426.081515] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[74426.083887] XPAD: original out.bInterval=8 -> new interval=8
[74426.083895] XPAD: original in.bInterval=1 -> new interval=1
[74426.084947] XPAD: original out.bInterval=8 -> new interval=8
[74426.084980] XPAD: original in.bInterval=1 -> new interval=1
[74426.086039] XPAD: original out.bInterval=8 -> new interval=8
[74426.086045] XPAD: original in.bInterval=1 -> new interval=1
[74426.087206] XPAD: original out.bInterval=8 -> new interval=8
[74426.087212] XPAD: original in.bInterval=1 -> new interval=1
[74450.430336] input: Xbox 360 Wireless Receiver (XBOX) as /devices/platform/soc/ffb40000.usb/usb1/1-1/1-1.1/1-1.1:1.0/input/input24
```

mihalski, 2020-01-26:
Hang on... Is the problem as basic as the Product ID not being in 360Controller/360Controller/Info.plist ???

FranticRain, 2020-01-26:
@mihalski There is a separate plist for wireless dongles: https://github.com/360Controller/360Controller/blob/master/WirelessGamingReceiver/Info.plist

mihalski, 2020-01-26:
@FranticRain Ah yes, so there is.. But the Product ID we are having an issue with also does not seem to be in there. Any chance it could get added for the next release?

FranticRain, 2020-01-26:
@mihalski Your logs report the device as: `idVendor=045e, idProduct=0291,`

That's:
VID: 0x045e
PID: 0x0291

Convert that from base 16 to base 10, so we can put it in the plist. That gives us:
VID: 1118
PID: 657

You'll notice that that is `WirelessGamingReceiverForWindowsAlternate`.